betaflight/betaflight-tx-lua-scripts #340: Stuck on "initialising"

JanJoh, 2020-05-11:
Taranis X9d+ ("Classic", not 2019).
Open tx 2.3.7

Soooo, i grabbed the nightly build as 1.4.1 did not work on my TX.  I wiped my SD Card and unpacked only the 26-version of the SD, card and the Nightly build. 

When selecting Betaflight in tool menu, it started to compile, and then dropped me out to the main menu. 

I chose the betafligth again, and got "Betaflight config Initialising", and it is stuck here. I left it for quite a few minutes without any change. 

Restarted the TX, and the same thing repeats.

klutvott123, 2020-05-11:
The compile part is normal. "Initialising" means it's waiting for a connection so that it can get the MSP version from your flight controller. Connect power to your quad and try again

RipperDrone, 2020-05-11:
verified BF 4.2 + Lua 1.5 Nightly 5 days agoon my Taranis X9D+ (2013) - worked like a charm, tested out in the field twice. I had to carefully delete all compiled .luac files in my Taranis folder, then install the nightly versions.

JanJoh, 2020-05-11:
> The compile part is normal. "Initialising" means it's waiting for a connection so that it can get the MSP version from your flight controller. Connect power to your quad and try again.
> We're going to change the message to something more explanatory.

THAT explains it. I was waiting for it to continue to some sort of "dataless" display like the old version i had. 

I feel sort of silly now.   I suppose i didn't interpret "initialising" as "waiting for data".. My bad.

closed 2020-05-11 by JanJoh

FPVRishi, 2020-11-24:
I seem to be stuck with the initialising msg too.

I am running the TBS nightly build on my xlite (2.3.10 opentx bootloader version) and sd card version 25.  

I got the the bf v1.5.0 Lia script files and on execute, the compile function ran, but now when I try the tools menu, I get a constant initial using msg, even when the quad is powered on. Any help appreciated...

klutvott123, 2020-11-24:
Stuck on "Initialising" indicates that it's not getting a response from the FC for some reason. Is telemetry working? Does it work with the official opentx 2.3.10?

FPVRishi, 2020-11-24:
Thank you for replying.  I've never used the bf Lua script before.  My Rx is a FrSky xm+ and I do get and can adjust the pid profile details on my goggles via bf osd.

So from what you are saying, maybe there's nothing wrong with my script setup and maybe I do not understand how to setup up the FC side of things well enough -i thought that it's just a case of setting up the script and running it, but I don't know if there's any additional setup required for the FC and Rx or soldering?

klutvott123, 2020-11-24:
You need a receiver with smartport telemetry support. R-XSR for example. The xm+ doesn't have telemetry. 🙂

FPVRishi, 2020-11-24:
Thank you so much - you have saved me hours of bewilderment.

I thought that as I was getting some form of telemetry back via channel 12 of my xm+ (seeing voltage, rssi, current on my bf osd), that it would work (thank you for correcting me).

I'm still learning and my next few builds should be r-xsr (and crossfire based-will the script work on this too?), so I'll be able to check to see if things are working then (but it would seem that everything may well be ok, except for my mistake in not understanding the Rx requirements!)

klutvott123, 2020-11-24:
You can get RSSI through a channel with the xm+, yes. The voltage and current you're seeing comes directly from betaflight.
The BF lua scripts works with crossfire too. 👍

FPVRishi, 2020-11-24:
That's what I've been doing with my xm+ buiilds. (getting rssi through a channel) and also using bf for voltage and current (

FPVRishi, 2020-11-24:
I really appreciate your help.  Thank you again.

slicerFPV, 2020-12-13:
I have the same problem. Smart audio is working in my goggles but not on my taranis lua scripts it’s stuck on initializing it’s working on my quad with tramp protocol. What could be the issue.

jooojo1988, 2022-01-13:
Same problem here. I have Edgetx running and receiving telemetry from betaflight, but BF lua 1.5 never gets passed "initialising"

klutvott123, 2022-01-13:
If you're using BF 4.3 you need the latest nightly https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases/tag/v20220111.5486 and be sure that your hardware meets the requirements detailed in the readme https://github.com/betaflight/betaflight-tx-lua-scripts#firmware-considerations

fdev31, 2022-01-19:
Same issue here, using:

- nightly Lua scripts
- ELRS 2.0.1
- Betaflight 4.3-rc2
- EdgeTX 2.5

klutvott123, 2022-01-19:
Is telemetry enabled and working?

haslinghuis, 2022-01-19:
@fdev31 please use latest nightly instead of RC2 as some issues were fixed.

fdev31, 2022-01-19:
That was telemetry not enabled, thank you so much !