betaflight/betaflight-tx-lua-scripts #342: [1.5.0] RSSI over MSP problem

0crap, 2020-05-12:
I installed the new 1.5.0 LUA scripts on my Taranis QX7S.
I tried the RSSI over MSP background option by running the lua background task.

It seems this is not fully reliable, in one flight I got three "RSSI LOW" warnings in my OSD.
After a replay of my goggle DVR I saw the RSSI icon was missing. I guess it was starting to blink because I have an alarm set for it in the OSD.
It was really short and reappeared very quick. (Within a second the RSSI icon was back again.)
The BBL log showed the real problem, RSSI was just not reported at all and dropped to the bottom.
Controls where still there, it's only RSSI that was not reported.

So maybe RSSI over MSP with the background LUA script is not the most reliable way to go?
[image]
[image]
[image]
[image]

klutvott123, 2020-05-12:
I have only tested it on the bench, but I don't think it's any less reliable than it was in earlier versions.
That it came back within a second sounds about right. RSSI is sent every 500ms and we know that data can get lost. You could try sending it more often by decreasing this value
https://github.com/betaflight/betaflight-tx-lua-scripts/blob/6a97e05c385d4ba93c5e37f7bc359b4bb48e46b7/src/SCRIPTS/BF/rssi.lua#L10
`INTERVAL = 25` would result in RSSI being sent every 250ms for example

0crap, 2020-05-13:
OK, I'll give 250ms a try and report back.

Before that a few questions:
- Do I need to "recompile" the script if I change this value? If yes, is there an easy way to trigger this besides delete everything on my SD card and start all over.
- Is it possible to disable the RSSI background task and let the script only do the BBL timestamp thing?
- What do you mean with "I don't think it's any less reliable than it was in earlier versions"? I might be missing something, but I was not aware this could be done with tx-lua-scripts in earlier versions.

Thx!

klutvott123, 2020-05-13:
No need to recompile. rssi.lua is a single file and small enough that it doesn't matter. A recompile can be triggered by changing this https://github.com/betaflight/betaflight-tx-lua-scripts/blob/6a97e05c385d4ba93c5e37f7bc359b4bb48e46b7/src/SCRIPTS/BF/COMPILE/scripts_compiled.lua#L1
back to `return false`.

The background script detects the RSSI source from the FC at startup. If the source is NONE or MSP it will send RSSI. If there already is a source like adc, rx channel, protocol it won't send anything.

RSSI through MSP has been available for a long time. For the telemetry version of the script it was done as part of the background function.

mikeller, 2020-05-13:
@0crap: Can you please post a `diff` of your flight controller configuration?
One thing to consider is that the RSSI transmission competes with telemetry for the small bandwidth available on the telemetry channel. So if your flight controller is configured to send a lot of metrics over telemetry it may be that there just isn't enough bandwidth left for the RSSI packets to get through.

0crap, 2020-05-13:
Sure thing!
And I think you will be surprised, I'm a geek that comments out telemetry stuff I don't need since a long time. (Still remaining from the time we had a lot of "telemetry lost" messages.)

```
# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.0 Apr 29 2020 / 08:08:49 (b480103d4) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF405STD, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF405STD
manufacturer_id MTKS
mcu_id 00<cut>
signature 

# name: iFlight iX5 V2

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource SERIAL_TX 11 A15
resource CAMERA_CONTROL 1 A08

# feature
feature -AIRMODE
feature LED_STRIP

# beeper
beeper -ON_USB

# serial
serial 0 2048 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led
led 0 11,15::AB:0
led 1 11,14::AB:0
led 2 4,14::AB:0
led 3 4,15::AB:0

# mode_color
mode_color 6 1 2

# aux
aux 0 0 0 900 1100 0 0
aux 1 1 1 900 1100 0 0
aux 2 13 2 1900 2100 0 0
aux 3 28 1 1400 2100 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWBAND  L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 3
vtxtable powervalues 14 20 26
vtxtable powerlabels 25 100 400

# rxfail
rxfail 3 h
rxfail 5 s 1000

# master
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 350
set gyro_notch1_hz = 315
set gyro_notch1_cutoff = 305
set dyn_notch_max_hz = 470
set dyn_lpf_gyro_min_hz = 0
set acc_calibration = -12,36,22,1
set baro_hardware = NONE
set min_check = 1015
set max_check = 1990
set max_aux_channels = 4
set dshot_idle_value = 500
set dshot_bitbang = OFF
set motor_pwm_protocol = DSHOT300
set vbat_max_cell_voltage = 420
set vbat_warning_cell_voltage = 340
set small_angle = 180
set yaw_deadband = 4
set pid_process_denom = 2
set telemetry_disabled_acc_x = ON
set telemetry_disabled_acc_y = ON
set telemetry_disabled_acc_z = ON
set telemetry_disabled_pitch = ON
set telemetry_disabled_roll = ON
set telemetry_disabled_heading = ON
set telemetry_disabled_altitude = ON
set telemetry_disabled_vario = ON
set telemetry_disabled_lat_long = ON
set telemetry_disabled_ground_speed = ON
set telemetry_disabled_distance = ON
set osd_warn_rssi = ON
set osd_rssi_alarm = 45
set osd_cap_alarm = 1200
set osd_rssi_pos = 2448
set osd_tim_2_pos = 2453
set osd_throttle_pos = 2082
set osd_current_pos = 2402
set osd_mah_drawn_pos = 2442
set osd_warnings_pos = 14667
set osd_avg_cell_voltage_pos = 2435
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set osd_stat_max_fft = ON
set debug_mode = RC_SMOOTHING_RATE
set cpu_overclock = 192MHZ
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 3
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5732
set vcd_video_system = NTSC
set camera_control_key_delay = 100
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_q = 700
set name = iFlight iX5 V2

profile 0

# profile 0
set dterm_lowpass2_hz = 250
set anti_gravity_gain = 4000
set p_pitch = 58
set i_pitch = 95
set f_pitch = 160
set p_roll = 46
set i_roll = 95
set f_roll = 160
set i_yaw = 120
set f_yaw = 130
set d_min_roll = 0
set d_min_pitch = 0
set ff_interpolate_sp = AVERAGED_3

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set rates_type = ACTUAL
set roll_rc_rate = 15
set pitch_rc_rate = 15
set yaw_rc_rate = 10
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 61
set pitch_srate = 61
set yaw_srate = 45
set tpa_breakpoint = 1300

rateprofile 1

# rateprofile 1
set rates_type = ACTUAL
set roll_rc_rate = 15
set pitch_rc_rate = 15
set yaw_rc_rate = 10
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 61
set pitch_srate = 61
set yaw_srate = 45
set tpa_breakpoint = 1300

rateprofile 2

# rateprofile 2
set rates_type = ACTUAL
set roll_rc_rate = 15
set pitch_rc_rate = 15
set yaw_rc_rate = 10
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 81
set pitch_srate = 81
set yaw_srate = 45
set tpa_breakpoint = 1300

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

klutvott123, 2020-05-13:
https://github.com/betaflight/betaflight/pull/9771 may also help

0crap, 2020-05-13:
I installed 4.2.0 RC2 and was able to reproduce this issue on the bench.
Changing rssi.lua to 250ms made a huge improvement!
But after waiting long enough I still see it happen on the bench.

I reverted to using RSSI over AUX4 which of course solved it.
Maybe it does not help I use SmartPort over SoftSerial, if that has anything to do with it.

mikeller, 2020-05-16:
@0crap:

> Changing rssi.lua to 250ms made a huge improvement!

So you changed from `local INTERVAL = 50` to `local INTERVAL = 250`?

0crap, 2020-05-16:
Correct, I did local INTERVAL = 25 as suggested by klutvott123.
I didn't try any shorter because I don't know what the negative effects can be of doing so.

As I now understand earlier lua versions worked similar for sending rssi, what time setting did earlier lua versions use?

klutvott123, 2020-05-16:
Earlier versions also used `INTERVAL = 50`

0crap, 2020-05-16:
Interesting, makes you wonder why it became less reliable.
I'm pushing my flying buddy to upgrade all his stuff so I can see if it's only me.
Guess not a lot use lua 1.5.0 yet, because you need OpenTX 2.3.4 or newer.
(Some I know use really old OpenTX and just don't care.)

0crap, 2020-05-16:
Since OpenTX playes a role in this, I'm using the TBS nightly that supports CRSFshot.
Looking at the timestamp it's just after the official 2.3.4 release (2019-12-23).
So I should be good. :-)

[image]

klutvott123, 2020-05-16:
New and old version does exactly the same. Send RSSI at a set interval and hope for the best 😄 
I can't see how any of them would be more reliable than the other.

If you're not seeing any "protocol not supported" messages then everything should be fine. 2.3.4 has improvements that let us reliably detect which telemetry protocol is being used.

0crap, 2020-05-16:
Nope, no "protocol not supported" messages, so that seems fine.

Personally I have no experience with older lua versions in regard to RSSI.
(I always used an AUX channel or protocols that support it.)
It's my first try with lua 1.5.0 to test this functionality.

However my flying buddy uses the old lua for a long time to give him RSSI inside the goggles.
He does have problems with it because it does not work always, but that must be the "hope for the best" part like you say. :-)
He will install lua 1.5.0 and test to see if the behaviour changes.
I'll report back on that later.

All I know is that on my rig it is not reliable enough to use it.
Like I described in my opening comment for this issue.
Would be nice if other users can comment on this issue.
Also if it works perfectly fine for them. 👍

klutvott123, 2020-05-16:
If he's not getting RSSI at all, it's most likely because the script is visible. The telemetry scripts background function is prevented from running when the script is visible. The new tools version is functionally the same because telemetry/function scripts are stopped when running a one time script. When exiting "Betaflight setup" the background script will be started again.

mikeller, 2020-05-16:
@klutvott123: We might want to consider decreasing the interval for RSSI, or make it switchable.

@0crap: I wonder if different firmwares on the CRSF TX / RX play into this - not sure if there are differences in how much bandwidth is available for telemetry, and how many telemetry packets end up being discarded.

0crap, 2020-05-16:
@mikeller To make sure there is no confusion, the quad in question is using SBUS with SmartPort over SoftSerial.
The RX is an R-XSR using the latest EU-LBT ACCST D16 v2.1.0 firmware.

I have other quads using CRSF, that's why I use the TBS OpenTX nightly.

And the idea to make the interval configurable for the end user does sound like a plan.

mikeller, 2020-05-16:
@0crap: Ok, I misread your previous statement then. But the same applies to FrSky protocols - the way queuing for telemetry packets works is not necessarily the same for all firmware versions.

0crap, 2020-05-19:
OK, tested with my buddies rig. Seems he is having a totally different issue.
One out of ten lipo plugins give him a blinking RSSI icon. If that happens it will not return the complete flight. After upgrading everything, that behaviour stays the same with the new 1.5.0 lua's. No change.
So that is unrelated to this issue where I always have RSSI on lipo plugin, but sometimes it's just not reported shortly during flight.

I guess make the INTERVAL somehow configurable for the end user might help in a few cases.
What I understand from klutvott123 is that RSSI over MSP will always be a bit "hope for the best".

closed 2020-06-28 by stale[bot]