betaflight/betaflight-tx-lua-scripts #343: TX hangs on 'Initialising' - Requires reboot

Hellfire82, 2020-05-13:
I'm new to lua scripts & it wasn't immediately obvious that the RX has to support telemetry in order for these scripts to work. Perhaps this could be added to the readme file to help beginners.

When attempting to run the 1.5.0 script with an XM+ (which doesn't support telemetry), the screen hangs on 'Initialising' & the TX has to be rebooted since no other buttons work.
Maybe a time-out period of 10 secs should be built in to prevent this from having to be done? I'm aware now that this feature can't be used & I therefore won't be encountering this issue again, however it is still an issue & should be fixed :)

FrSky XM+
OpenTX 2.3.7
Betaflight 4.2 RC1
LUA script 1.5.0

klutvott123, 2020-05-13:
No need for a reboot. Long press exit will stop the script

Hellfire82, 2020-05-13:
Ah so it does.
Thanks for the help & sorry for being a noob :)

PR #344 (2020-05-16): Added requirement for SmartPort to README.
Fixes #343.
closed 2020-05-13 by Hellfire82