betaflight/betaflight-tx-lua-scripts #345: Feature Request: OSD Warnings/Messages in Telemetry Pages with Audio

brunoleaomaia, 2020-05-17:
Currently it is not possible to receive the warning messages (warnings,  arming status flags, sanity checks, etc)  in DJI FPV System custom OSD.

An alternative for users with DJI FPV System or without OSD would be to receive these messages on the telemetry page (If possible with a voice message related to the message).

Today I'm using DJI FPV System with GPS Rescue but it is very difficult to know for example why my quad is not arming (SANITY CHECKS). It is very difficult to know if GPS Rescue is available during the flight because we did not receive the message GPS RESCUE N / A for example.

It would be great to hear a voice saying "GPS Rescue Not Available" and "GPS Rescue Restored" when you have your goggles on your face.

Thank you!
@maiafpv
https://instagram.com/maiafpv/

mikeller, 2020-05-17:
Ironically, this would probably best be added as a telemetry script. ;-)

Another option would be to cram these values into a 'normal' telemetry value, like we are already doing for flight modes with T1 - this could then either be consumed by a lua script, or potentially just used to trigger voice messages through a logical switch in the TX.

brunoleaomaia, 2020-05-17:
Yes, as I said at the beginning, on the 'Telemetry page' (I don't know if the correct term is this), but a numeric id would be sent via telemetry for example that represents a specific message, and a voice could be triggered via lua as you said.

This will help many pilots who do not have or do not use OSD for some reason and also who use DJI FPV System.

haslinghuis, 2026-11-09:
Closing as out of scope for this repo. These scripts are configuration menus, and the flight controller currently doesn't send warnings or arming-disable reasons over CRSF/SmartPort telemetry, so there is nothing for a Lua script to read or announce. This would first need firmware support for sending warnings as telemetry, which could then drive a telemetry script or a logical switch on the radio. Please open a feature request in the betaflight firmware repo if you'd like to pursue it.

closed 2026-11-09 by haslinghuis