betaflight/betaflight-tx-lua-scripts #348: Current meter offset [mA] - no negative values possible

RipperDrone, 2020-05-22:
Lua 1.5 release, latest BFC 10.-7, BF 4.2RC1 --> I can enter/load negative current offsets in BFC, however BF Lua script only shows positive (integer?) values. Whenn I roll down below zero, it reads 655xx, and when I save these values, current meter on OSD shows some 230+ Amps. 

Looks like unsigned integer vs. positive/negative values needed bug...

[Toothpicks - BetaFPV Knight 4 Betaflight Diff.txt](https://github.com/betaflight/betaflight-tx-lua-scripts/files/4669193/Toothpicks.-.BetaFPV.Knight.4.Betaflight.Diff.txt)

RipperDrone, 2020-05-22:
@klutvott123 pls explain: isn't it the lua script formatting the data received from the FC into the edit fields disaplyed on the RC? When using the OSD to display/change current offset, it shows negative offsets correctly. However, having data displayed on the RC screen and decrementing from "0", huge positive numbers (around 655xx) are shown instead of negative ones.

klutvott123, 2020-05-22:
No, it's OpenTx. BF lua has nothing to do with the built in OpenTx telemetry screens

RipperDrone, 2020-05-22:
ok, thank you. Will check w/ OTX then...

closed 2020-05-22 by RipperDrone

RipperDrone, 2020-05-22:
@klutvott123 sry, I re-edited the issue description. Issue is not related to BF lua, as you said, however it is precisely about NOT being able to set negative current offset values on BF OSD. (I was wrong in my initial description, it didn't happen with lua script...). Thus, it does not seem OTX related, but BF related then, correct? If yes, I can open an issue ticket there...

RipperDrone, 2020-05-22:
current.c: Doesn't this cast have to be uint32_t instead of int32_t maybe?
[image]

klutvott123, 2020-05-22:
@RipperDrone Yes, looks like it's in betaflight. I can reproduce in configurator when I set a negative offset, but the sensor in Opentx shows 0.0. Which telemetry protocol are you using?

RipperDrone, 2020-05-22:
@klutvott123 Using FPORT. Have created an issue in BF...