betaflight/betaflight-tx-lua-scripts #349: Issue gettin LUA scripts on QX7

Dyrt-Nasti, 2020-05-22:
Just bought Taranis Qx7, installed all new firmware: OpenTX-x7access version 2.3.7, installed LUA Scripts 1.5.0.. Managed to write scripts to the taranis, hold down on "PAGE" tab and "Use tool menu instead" pops up. I go to tool menu, to the betaflight configurator. Says "Initializing"
I plug in the quadcopter, gain telemetry but "Initializing" screen doesnt go away and still cant access LUA Scripts

klutvott123, 2020-05-22:
It's not getting the MSP version from the FC for some reason. Which Betaflight version? Which receiver/firmware version? Are you getting telemetry from the FC or just RSSI/voltage sensors from the receiver? Have you done "discover sensors" in OpenTx?

Dyrt-Nasti, 2020-05-22:
i got betaflight 4.1.6 on my quad with an r-xsr reciever, im trying to find the firmware for the receiver.
I am getting telem. When i unplug my quad my taranis says telem. lost. And yes i did discover new sensors

Dyrt-Nasti, 2020-05-22:
i think i have to update the firmware on my receiver. i just broke my last and got a new one and didnt flash new firmware

Dyrt-Nasti, 2020-05-23:
the receiver wont flash new firmware, I tried doing it through the taranis but I kept getting "FM Update Error, Not Responding". Then i tried serial passthrough and i couldnt get the files onto the receiver.

klutvott123, 2020-05-23:
I'm not too familiar with access, but doesn't it support OTA firmware update? Did the lua scripts work with your last receiver? Which FC do you have?

Dyrt-Nasti, 2020-05-24:
I didnt have lua scripts on my last receiver, this is my first try. I have the iflight F405 FC and i think it does support OTA

klutvott123, 2020-05-24:
Make sure to select "Flash S.Port" and not "Flash external device" when doing it through the TX

sriharis123, 2020-06-28:
I don't know if you've solved this, but I just ran into the problem as well (also qx7/r-xsr). Fixed it by upgrading the firmware on my XJT module.

fabioraphael, 2020-07-31:
Same Here, "Initialising"
First time started to compile, now in TOOL menu, Betaflight config "Initialising", and it is stuck.

QX7 ACCST
Firmware tbs_firm_x7_en_20200106
BF 4.2
Crossfire 3.72
LUA Scripts 1.5.0.

Anyone can help me?

Thanks

klutvott123, 2020-07-31:
@fabioraphael Is your quad powered on?

fabioraphael, 2020-07-31:
> @fabioraphael Is your quad powered on?

Yep! w/ Telemetry

klutvott123, 2020-08-01:
@fabioraphael Then the only thing I can think of is to try "Discover new sensors". I'm not familiar with the tbs stuff, but I know there are people using the lua scripts with their opentx builds.

fabioraphael, 2020-08-01:
> Then the only thing I can think of is to try "Discover new sensors". I'm not familiar with the tbs stuff, but I know there are people using the lua scripts with their opentx builds.

I am tray it too but stuck in Initialising

klutvott123, 2020-08-06:
@fabioraphael Then msp over telemetry isn't working for some reason. It could be a RX/TX issue. is 3.72 the latest crossfire firmware?

fabioraphael, 2020-08-07:
Solved, Many Tks.
Wire Wrong  TX1 Invertor.

> @fabioraphael Then msp over telemetry isn't working for some reason. It could be a RX/TX issue. is 3.72 the latest crossfire firmware?

DavidFpv, 2020-08-28:
@klutvott123 
This problem also calls for me ...
How did you solve it?
my setup:
x9d fw 2.3.9
lua 1.5.0
bf fw 4.2.0

That I enter "TOOLS" I get "Initialising" and nothing changes.
When the drone with battery plag in, and telemetry works...

klutvott123, 2020-08-28:
@DavidFpv Double check wiring, update RX/TX firmware, "Discover new sensors". Which receiver do you have?

DavidFpv, 2020-08-29:
@klutvott123 
The cables are connected correctly. 
Otherwise I would not see the remote in a configurator or telemetry. 
The xjt on 2.1.0 The rx on 2.1.0 And I was looking for new sensors there are only 2 Battery voltage and rssi,
My rx is r xsr

klutvott123, 2020-08-29:
If those are the only sensors you see then telemetry isn't working correctly. Is it Fport or SBUS/smartport?

DavidFpv, 2020-08-29:
@klutvott123 
SBUS and smartport...
I  have thus bord:
https://www.copterfarm.de/media/pdf/3d/bb/f7/RDQ-Bardwell-F4-FC-Manual-v2-1.pdf
[image]
[image]

DavidFpv, 2020-08-29:
By the way I tried with a drone on CROSSFIRE and also with it it did not work ...
And there I get all the sensors

klutvott123, 2020-08-29:
Did you enable telemetry and configure a uart3 for smartport? did you do ```set tlm_halfduplex = off``` as instructed in the manual? 

Are you using the latest crossfire firmware? There have been bugs in older firmwares that prevented msp over telemetry from working properly. Which Betaflight version and FC is it?

DavidFpv, 2020-08-29:
on the drone with rxsr - I dont did (set tlm_halfduplex = off) What it means?

on the drone with crossfire - I do not remember but I will check in a few more minutes

BF FW bord with:
rxsr- 4.2.1-2 will check in a few more minutes
cross- 4.2

DavidFpv, 2020-08-29:
@sriharis123 
Okay I saw what you mentioned and I wrote it in CLI now I see a lot more sensors.
And the LUA works thank you very much !!!!
Now just what's the problem with CROSS

crossfire firmware is 3.72 tx lite
It's normal just without Bluetooth

sriharis123, 2020-08-29:
Awesome! Good to hear

DavidFpv, 2020-08-29:
@sriharis123 
Yes but the skimmer with CROSSFIRE still does not show LUA

klutvott123, 2020-08-29:
@DavidFpv Read further up. @fabioraphael had a similar issue where crossfire with telemetry was working but lua was not. Turned out to be a wiring issue in his case.

DavidFpv, 2020-08-29:
@klutvott123 Can you please send me a link to the forum?
I'm new here and have not found the subject

klutvott123, 2020-08-29:
Here https://github.com/betaflight/betaflight-tx-lua-scripts/issues/349#issuecomment-666907769

DavidFpv, 2020-08-29:
> Then msp over telemetry isn't working for some reason. It could be a RX/TX issue. is 3.72 the latest crossfire firmware?
@klutvott123 
What do you mean here?

klutvott123, 2020-08-30:
@DavidFpv msp is a protocol used to communicate with the flight controller. The Betaflight lua scripts uses this protocol over the telemetry link to communicate with the flight controller. The lua scripts sends a request to the flight controller which then sends the requested data back. For this to work we need communication both ways. Your issue seems to be that you're not getting communication both ways. Could be caused by a wiring or configuration error.

DavidFpv, 2020-08-31:
@klutvott123 
I get all the sensors so there is a two way connection ...
Maybe something in beta?
Because I did not set anything special for LUA

klutvott123, 2020-08-31:
@DavidFpv Could be anything. Which FC and how is it wired?

DavidFpv, 2020-08-31:
@klutvott123 
https://www.google.com/search?q=Original+Airbot+Omnibus+F4+V5+Flight+Controller&safe=strict&source=lnms&tbm=isch&sa=X&ved=2ahUKEwiUpdHH2MXrAhWBUcAKHZxhAjkQ_AUoAXoECAsQAw&biw=1920&bih=937#imgrc=X5F8BGvI9yW-eM

rx ch1 (crsf tx) to rx6 on the bord
rx ch2 (crsf rx) to tx6 on the bord

ksumit12, 2020-09-03:
@DavidFpv  how did you solved your problem with r xsr drone!!
I am facing same from from with my drone just upgraded firmware on qx7 to opentx 2.3.9 and added lua script 1.5.0
Since then I am facing this issue 
And yes  all my other sensors are working rssi,rxbat and vfas .

DavidFpv, 2020-09-05:
@ksumit12  I had to write something in cli designed for my fc. (Fc: jbf4) And after I wrote that everything worked.
@fabioraphael or @klutvott123  You can try to help me solve the problem with the crossfire?

DavidFpv, 2020-09-05:
@ksumit12 try to update version of your Xjt and rx...

ksumit12, 2020-09-05:
@DavidFpv yeh I tried to upgrade my rx firmware to latest 2.x.x but that version is faulty it won't bind to my tx after upgrading I m glad it didn't bricked my rx as this version is bricking rx for some people 
And I also tried to upgrade internal xjt module still no luck
Also by reading other threads I tried to format my tx sd card and again pasting the data but still nothing works for me

klutvott123, 2020-09-12:
@DavidFpv That wiring should be fine. According to the manual any inverter is bypassed when connecting directly to rx/tx6. Then it has to be something else. Hard to say what.

DavidFpv, 2020-09-12:
@klutvott123 So what can I do?