betaflight/betaflight-tx-lua-scripts #353: VTX tables use model name instead of craft name

kristjanbjarni, 2020-06-08:
The VTX tables use the model name from the TX instead of using the craft name from BF. That means when you have two or more crafts with the same model in the TX the VTX settings page uses incorrect values for the other crafts.

To reproduce the problem: Create a model with some name in the TX that controls two different quads with different VTX hardware and values. When running the script only the first quad VTX tables are downloaded.

This is running on OpenTX 2.3.7 with Lua scripts 1.5.0 on Jumper T16

mikeller, 2020-06-09:
@kristjanbjarni: This is by design. Just copy the models on your TX, and use different names for the individual craft - using the craft name is by no means guaranteed to work better, as a lot of people don't even bother with setting one, and there is no way to make sure they aren't using the same name on multiple craft (like when they are using it for the pilot name).

kristjanbjarni, 2020-06-10:
Well that is unfortunate because everybody in this hobby who owns a couple of aircrafts with the same protocol it is customary to use the same model for all the aircrafts to minimize duplicating work for model changes. But anyway for people who don't use the craft name there can just be a fallback to the model name if craft name is empty, and since there is a different variable for pilot name, it really shouldn't be used for pilot name at all.

But if this is the case then I will probably just make my own version of the lua script to handle different craft names.

kristjanbjarni, 2020-06-22:
Since this is by design I made a modified version that supports using custom craft name for the vtx settings. I uploaded my changes to github since others can probably benefit from this. If you want to use some or all of  my changes for the betaflight-tx-lua-scripts you are very much free to do so.

Here is my repository: [CraftName](https://github.com/kristjanbjarni/CraftName)

closed 2020-06-22 by kristjanbjarni