betaflight/betaflight-tx-lua-scripts #357: BF settings script significantly faster value loading times on X9D+ than Jumper T16

fiechr, 2020-06-27:
The setup is a Tinyhawk Freestyle II (BF 4.1.7) with a FrSky R-XSR receiver, SmartPort connected to the TX1 pad (configured as SOFTSERIAL 1).

R-XSR uses Firmware LBT 2.1.0, also X9D+ on XJT LBT 2.1.0 and Jumper T16 with the latest version of the Multiprotocol firmware (using the FrSkyX2 EU/LBT protocol).

I'm always running the scripts on both transmitters using the main menu, as recommended. Both use the latest stable release 1.5.0 of the Lua scripts and OpenTX 2.3.9

Receiving regular telemetry values seem to work fine with both transmitters. Not tested in detail, but at least the values show up and are regularly updated.

**The basic issue is:** While going through the different pages of the menu, the X9D+ loads the current values in about 1-5 seconds, while the Jumper T16 takes at least 20 seconds, often 1-3 minutes or more(!). Also downloading the VTX tables takes significantly longer on the T16 (about 90 seconds or more).

There are no other configuration changes, just binding on of the transmitters to R-XSR.

Any ideas why that is? How can I debug this to find out what's going on here?

mikeller, 2020-06-28:
@fiechr: One more thing to test: Plug a multiprotocol module into the X9D+, or an XJR module into the T16. My suspicion is that this might be caused by the different TX modules - the rest of the setup is pretty much identical.

fiechr, 2020-06-28:
Unfortunately, I do not have external XJT or MPM available to test this.

closed 2020-08-08 by stale[bot]