betaflight/betaflight-tx-lua-scripts #358: no pids values loaded

nangel0, 2020-06-30:
I have this odd situation delayed for a while and now I willing to sort it out.
I have a slim+ binded on 868lbt to r9(old) on RM Tx16s all wired to a kakute f4 using 2 wires via s. port and sbus as user manual says. Latest everything, BF 4.2, latest opentx, latest lua available and latest fw on r9 and slim+.
Problem: I have telemetry values on the radio, but BF lua doesn't work, it doest load any values, and stucks on vtx tables. if i create the vtx tables file manualy, it loads but no values are load t pids page.

Messing with half duplex and telemetry inversion doesn't change anything.

Everything works fine on another quad using jhemcu f4 and r9mm

mikeller, 2020-07-06:
@nangel0:

> Problem: I have telemetry values on the radio,

If you have telemetry values then the connection between Betaflight and OpenTX is working - no point changing anything there.

Question: What happens if you do 'Detect sensors' again?

klutvott123, 2020-07-06:
Maybe this is related to #357? Does the script get past the "Initialising" stage? "Downloading VTX Tables" always takes a while.

nangel0, 2020-07-21:
> Question: What happens if you do 'Detect sensors' again?

It detects the sensors again, but, something strange happens: the * blinks during some seconds and after a while it warns sensor lost and some off them (sensors) don't update anymore. 

> Maybe this is related to #357? Does the script get past the "Initialising" stage? "Downloading VTX Tables" always takes a while.

Not the same issue, it never loads...

klutvott123, 2020-07-21:
You said it loads if you create the vtx tables file manually. That means it got past the "Initialising" stage and that msp over telemetry is working.

nangel0, 2020-07-21:
I didn't explained properly. Sorry. 
The script always  hangs loading vtx tables and doesn't shows anything else.

If I create a vtx lua file I passes the loading vtx screen but doesn't loads any pid values on any screen...

nangel0, 2020-08-02:
F.Port solved

RX1 from SLIM+ (flashed to 190201 FW) to TX3 from kakute F4
then 
CLI
serialrx_inverted = OFF
serialrx_halfduplex = ON

closed 2020-08-02 by nangel0