betaflight/betaflight-tx-lua-scripts #360: Horus X10S

4dimensio, 2020-09-05:
Good afternoon,
 I have a Horus X10S running OpenTX 2.3.5 and trying to use BF lua 1.5.0. I go to System>Tools> and click Betaflight Setup which is #356 9 in my tools menu and all that happens then on screen is " Betaflight initializing" that's it. I have even went into the "scripts" folder and clicked the scripts themselves whether it's PIDS,vTX lua scripts and nothing. The screen will flash and that's it. Am I missing something or do I need to use a lower BF lua scripts like 1.4.1?

Thank you for your time

klutvott123, 2020-09-12:
Don't you need the TBS nightly build of OpenTX to use CRSFshot?

4dimensio, 2020-09-13:
My issue has been already addressed,thank you for answering back.
 The post can be closed.

closed 2020-09-13 by 4dimensio