betaflight/betaflight-tx-lua-scripts #371: Initializing problem with FPort

GeraldoGarcia, 2020-11-07:
Frsky R-XSR receiver flashed with the latest FPort ACCST firmware (v2.1.0) running through the univerted pad.
Taranis X9D+SE ACCST with OpenTX 2.3.10 (FW-XJT-v2.1.0) 
HGLRC XJB F428 stack: F405 flight controller and TX20 v2 VTX (tramp)
Betaflight 4.2.4, Betaflight Lua Script 1.5.0

I am posting it here because it may be an issue between the script and FPort and also because I could ind info about this anywhere else.
  
When I try to run the "Betaflight Setup" under "Tools" it goes to "initializing", then to "Downloading VTX Tables" and stucks. Never goes beyond that. 

VTX table manually configured inside BF, tramp working just fine as I can change the VTX settings on the OSD. Smartport telemetry also working (15 sensors reported)  

I have another quad running an R-XSR with SBUS and the script works just fine on that one. Is there an issue with FPort or am I missing something? On RC Groups someone said FPort and Betaflight Lua don't wok well together, but i suppose it should work.

Thanks for any insight you could spare. 
 
[BTFL_cli_AeroZero_XJB145_20201106_233447.txt](https://github.com/betaflight/betaflight-tx-lua-scripts/files/5504187/BTFL_cli_AeroZero_XJB145_20201106_233447.txt)

klutvott123, 2020-11-07:
It just takes a little longer with Fport and downloading the vtx tables can sometimes take a minute or more.
Disabling any sensors you don't need in cli could possibly speed up things a little.
There were some issues with both Fport and sbus/smartport in the past, but with the 2.1.0 firmware everything works as it should.

GeraldoGarcia, 2020-11-07:
Thanks for the info, it kind of worked but not at first.
I disabled the unused sensors on CLI, set the VTX on pit mode to avoid overheating and tried to run the script. It stayed in "downloading vtx tables" for way more than 5 minutes and I interrupted it.
As a last try before swapping the receiver, I deleted and reinstalled (again) all the SD card contents for the radio and BF LUA scripts, turned off all 2.4ghz devices on the house.
I plugged a full battery on the quad, started the "Betaflight setup" script and was prepared to let it run for as long as possible when I remembered I had not set the VTX to pit mode. While the scrip was still "downloading vtx tables" I entered the osd using stick commands to enable pit mode. As soon as I set the VTX to pit mode and removed the googles the scrip menu was there. It took less than 2 minutes total.
Was it coincidence? The reinstalled SD card and scripts? The usage of the 'tramp' during the download unlocked something? Cannot say, but it is working now. 
Thanks.

closed 2020-11-07 by GeraldoGarcia