betaflight/betaflight-tx-lua-scripts #372: No connection - "Initialising" (Xlite pro, R9mm, F405 Mini)

maxledfreak, 2020-12-09:
1. Describe your problem
Transmitter shows "Initialising" after power on my FC. 
2. Include ways to reproduce the problem;
Problem occurs each power cycle. FC is receiving signals without issues. Transmitter is receiving signals from FC/receiver(telemetry).  --> general TX and RX is working.  
BetaFlight configuration:
UART1 --> Serial Receiver
Protocol --> FrSky f.port
telemetry --> active
HW/SW: 
FC: Mamba F405 Mini MK2 with BF: 4.1.1
Transmitter: FrSky Xlite pro (SW: 2.3.10 opentx) + R9m lite pro (SW: 1.1.0/1.3.0FLEX) 
Receiver: R9mm OTA (SW: 1.1.1/1.3.1)
Betaflight TX Lua Scripts 1.5.0
Wiring:
FC --> RX:
5V --> V_IN
GND --> GND
SBUS --> Inverted S.Port

Thanks for support.
Max

mikeller, 2020-12-09:
@maxledfreak:

> FC: Mamba F405 Mini MK2 with BF: 4.1.1

You'll have to update to Betaflight 4.2 to use the latest lua scripts.

maxledfreak, 2020-12-09:
Same issue with BF 4.2.2. Downward compatibility should be given, I hope so.

klutvott123, 2020-12-09:
Could be a fport firmware bug. Can you try with SBUS/smartport?

> SBUS --> Inverted S.Port

This doesn't look right unless you did some remapping. Does the SBUS pin have an inverter? If so maybe you should try using the regular SPORT pin on your receiver.

maxledfreak, 2020-12-09:
I will try first setup with SBUS/smartport.

Regarding your second point. Maybe it is a little bit confusion with inverted and non inverted. "S.Port/F.Port" (exact wording on RX) does not work. Inverted S.Port works (telemetry and control) but without lua script.

bauweraertserwin, 2020-12-10:
Same problem here,  BFL 4.2.5 XLITE pro and f.port.

klutvott123, 2020-12-10:
@bauweraertserwin Which receiver and firmware version? Is telemetry working? Are you getting all sensors or just the ones from the receiver?

mikeller, 2020-12-11:
@maxledfreak:

> Downward compatibility should be given, I hope so.

Full downward compatibility is achieved by installing the lua script version that was released with the firmware version. lua on the TX is extremely constrained in memory, so besides some simple additional fields that are shown or not, backwards compatibility across protocol changes is not possible.

bauweraertserwin, 2020-12-14:
@klutvott123 
Rx = R9 MM - OTA (access) (F.port) v. 1.3.1
Telemetry is working 
BFL 4.2.5

klutvott123, 2020-12-14:
@bauweraertserwin Not getting past "Initialising" indicates that telemetry isn't working as it should. Could you guys please try version 1.4.1 of the scripts and see if any of the pages will load?

bauweraertserwin, 2020-12-16:
Version 1.4.1 gives a Syntax error

klutvott123, 2020-12-16:
1.4.1 can't run from the tools menu. It has to be setup as a telemetry screen

bauweraertserwin, 2020-12-16:
Ok i have the pids working.  The other telemetry screens are not working.  Says no telemetry  and gives a memory error when browsing through the different screens.

klutvott123, 2020-12-16:
Good. The memory error is just because it's compiling the pages as they are loaded. If you reboot your radio it will probably go away since the pages will already be compiled. 

Does it say "no telemetry" on the working pid page too? 
Try setting any of your pid values to 125 or 126 and save the page. Does it still work?

bauweraertserwin, 2020-12-16:
Yes it keeps working after pid 126,  and yes no Telemetry keeps flashing on the Telemetry screen.

klutvott123, 2020-12-16:
All right. then I assume 125 worked too. "no telemetry" means RSSI is either 0 or the sensor isn't discovered in opentx. Try "discover new sensors" on the telemetry page. Also provide a diff please.

bauweraertserwin, 2020-12-16:
It's working now.  Thank you very much for all the help! 

# 

# Diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.5 Nov 22 2020 / 18:39:53 (afdac08b3) MSP API: 1.43
# config: manufacturer_id: JHEF, board_name: JHEF7DUAL, version: 6fac4231, date: 2019-10-18T22:07:46Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name JHEF7DUAL
manufacturer_id JHEF
mcu_id 004300353237510731313935
signature 

# name: DIATONE ROMA F5

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY

# beacon
beacon RX_LOST
beacon RX_SET

# map
map EATR1234

# serial
serial 1 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 900 1200 0 0
aux 2 2 1 1300 1700 0 0
aux 3 46 2 1800 2100 0 0
aux 4 41 3 1600 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 25 200 600 800
vtxtable powerlabels 25 200 600 800

# master
set gyro_lowpass2_hz = 500
set dyn_notch_width_percent = 0
set dyn_notch_q = 300
set dyn_notch_max_hz = 500
set dyn_lpf_gyro_min_hz = 400
set dyn_lpf_gyro_max_hz = 1000
set acc_trim_pitch = -4
set acc_trim_roll = -2
set acc_calibration = 16,38,67,1
set align_mag = CW180FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 1800
set baro_hardware = NONE
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1400
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_descent_dist = 100
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 2
set osd_vbat_pos = 2530
set osd_rssi_pos = 2490
set osd_vtx_channel_pos = 2464
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2550
set osd_gps_speed_pos = 57
set osd_gps_sats_pos = 26
set osd_home_dir_pos = 2140
set osd_home_dist_pos = 121
set osd_battery_usage_pos = 2475
set vtx_band = 1
set vtx_channel = 4
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5805
set gyro_1_align_yaw = 900
set gyro_2_align_yaw = 900
set name = DIATONE ROMA F5

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 125
set dyn_lpf_dterm_max_hz = 305
set dterm_lowpass2_hz = 270
set anti_gravity_gain = 4000
set feedforward_transition = 100
set iterm_relax_cutoff = 10
set throttle_boost = 0
set p_pitch = 64
set d_pitch = 50
set f_pitch = 0
set p_roll = 59
set d_roll = 46
set f_roll = 0
set p_yaw = 63
set f_yaw = 0
set d_min_roll = 0
set d_min_pitch = 0
set auto_profile_cell_count = 6

profile 1

profile 2

# restore original profile selection
profile 1

rateprofile 0

# rateprofile 0
set roll_rc_rate = 65
set pitch_rc_rate = 65
set yaw_rc_rate = 55
set roll_expo = 7
set pitch_expo = 7
set yaw_expo = 7
set roll_srate = 74
set pitch_srate = 74
set yaw_srate = 74
set tpa_breakpoint = 1400
set throttle_limit_percent = 85

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

klutvott123, 2020-12-16:
@bauweraertserwin Great! Then 1.5.0 should work too. Thanks for testing 👍

bauweraertserwin, 2020-12-16:
I just tested the 1.5 with a clean sd card,  including the background script on special functions.  It says  downloading vtx tables,  and than it hangs.  So i had to revert to 1.4.1

klutvott123, 2020-12-16:
@bauweraertserwin I would recommend trying with 1.5.0 again. Downloading vtx tables can take a while in some cases especially when using fport.

bauweraertserwin, 2020-12-16:
Ok I'll try again.

bauweraertserwin, 2020-12-21:
Everything is working perfectly now.  I was a little too impatient.  Thank you again for helping me out!

NightWatchman, 2020-12-26:
I seem to be having this issue as well.
R-XSR receiver, v1.9 firmware with ACCST protocol, using smart-port telemetry
JBF7 flight controller, with betaflight v4.2.5

My Jumper T-16 radio sees telemetry alright from both the flight controller and the receiver, but running Betaflight configuration from the tools menu just hangs on "Initializing".

In a week or so I'll try again after flashing my receiver to v2 using the ACCESS protocol. I can't do it until I update the firmware in my radio's internal multi-protocol module.

klutvott123, 2020-12-26:
@NightWatchman If its firmware version 191128 it should work, but it's a good idea to update the TX/RX firmware to the latest versions anyway. In some cases people have fixed it by just updating the multiprotocol module. Also try "Discover new sensors" again.

NightWatchman, 2020-12-26:
Ya I'm using 191128 on my R-SXR. I'll keep poking at it and update my module when I can.

bobbillfold, 2021-01-13:
I am have an issue similar to these. I'm stuck on initializing. I think I read this thread a dozen times and believe I understand that Betaflight Lua is written for certain rx protocols? I'm using ExpressLRS. I assume this is the issue? Is it supported? Or will it be? Thank you

mikeller, 2021-01-13:
@bobbillfold: What telemetry protocol does ExpressLRS use?

bobbillfold, 2021-01-13:
@mikeller: CRSF, but it could be that they haven't supported full telemetry yet. A work in progress.

I paused and went to their discord. Got answered quickly. They haven't written full telemetry into ExpressLRS yet. PRs are in. Guess I'll be waiting patiently. Thanks for your time and all you do.

closed 2021-02-20 by github-actions[bot]

jianwwpro, 2022-05-29:
STM32 F405 + elrs 2.4G 
GHF420AIO 
same problem.  open betaflight config "Initialising"