betaflight/betaflight-tx-lua-scripts #373: BF 4.2 lua 1.5 working on Tramp protocol but not SA 2.0

slicerFPV, 2020-12-13:
Using BF 4.2 with The 1.5 lua on my x7 it’s working fine with my Tramp vtx but not with my Foxeer clear tx2 which uses tbs smart audio 2.0 unlocked. When I look in the goggles it’s getting all my smart audio information just fine so I know my vtx is broadcasting smart audio data but on my taranis it stays stuck on initializing what could be the issue.

mikeller, 2020-12-13:
@slicerFPV: You are mixing different things here - SmartAudio does not relate in any way to the Betaflight TX lua scripts - the data required for it cannot be transmitted over the VTX signal.
Please add details of what TX / RX hardware, protocols, firmware versions you are using on the problematic craft.

slicerFPV, 2020-12-13:
I use to be able to use smart audio or tramp to get my pids and vtx info on my taranis radio using smart port. Is that not available anymore? Cause all of my tramp vtxs are still allowing that thru the tools page and lua script on my taranis.

klutvott123, 2020-12-13:
If it's stuck on "Initialising" it means that telemetry isn't working properly. We really need the information that @mikeller asked for in his post above to be able to help.

slicerFPV, 2020-12-13:
My rx is frsky x4r-sb I’m not sure how to check the firmware on that. And my Tx is the Frsky Taranis qx7 on Open tx 2.3.10 with the open tx 1.5.0 betaflight lua scripts. I’m running a Foxeer Clear TX2 vtx with smart audio 2.0 unlocked. And my flight controller is CLracing f4 with betaflight 4.2.5. My craft is transmitting the smart audio data to my osd in the goggles just fine just not to my radio which just stays on initializing. My other crafts with tramp protocol are showing on the radio just fine.

klutvott123, 2020-12-13:
Then I assume you're using SBUS/Smartport. Are you getting telemetry from FC to TX at all?(check telemetry page in opentx).
Make sure that both SBUS and smartport wires are connected to the appropriate pins on your flight controller and that betaflight is configured correctly. telemetry enabled, uarts configured etc.

slicerFPV, 2020-12-13:
Yes I’m using sbus smart port. I definitely have the wires soldered up right cause everything was working with the old lua scripts on the older firmware back when you use to run the lua scripts as a telemetry page. And yes I’m getting all the telemetry like like vbat rssi etc.

klutvott123, 2020-12-13:
It could be a rx firmware bug. Several of frskys receivers have had a bug that caused problems with telemetry. This wasn't as noticeable with regular telemetry but it caused a lot of problems when using lua scripts. It has been fixed for some receivers but I don't know about the x4r-sb specifically. I would recommend updating to the latest firmware for your receiver and try again.

slicerFPV, 2020-12-13:
I’ll do that and see what happens. Maybe that’ll work.

slicerFPV, 2020-12-14:
Ok iv updated everything, my receivers my qx7, my flight controller and my vtx to the latest firmware and still having no luck. Every vtx that I have running smart audio instead of tramp protocol is staying stuck on initialising.....

klutvott123, 2020-12-14:
It just sounds very strange. The communication between your quad and radio has nothing to do with the vtx protocol. You don't even need to have a vtx connected for the lua scripts to work. Is the vtx the only difference between your working and non working quads? Are they all using the exact same receiver with the same firmware?

slicerFPV, 2020-12-15:
Working quads
-Tramp HV vtx, frsky x4rsb receiver.
-Tramp HV vtx, frsky rsxr receiver.
Non working quads
-TBS unify Evo vtx on Smartaudio 2.1, frsky rsxr receiver.
- Foxeer ClearTX2 vtx on Smartaudio 2.0, frsky x4rsb receiver.

I also have one quad with a frsky r9mini receiver that’s on tramp that won’t work either but its never worked on the r9 ever since I switch the quad to the r9. 
All receivers are on the most current firmware as of yesterday but still the same problem. My taranis OS and internal XJR module are also on most recent firmware.

slicerFPV, 2020-12-15:
Fixed Issue was with the smartport connection and the tlm_halfduplex setting. The board reset after flashing and these settings needed to be turned back off.

closed 2020-12-15 by slicerFPV