betaflight/betaflight-tx-lua-scripts #376: TBS Tracer, LUA stuck at "initializing"

oyvinla, 2020-12-29:
I'm trying to get the Betaflight Lua script V 1.5.0 to work, but it doesn't seem to work with TBS Tracer (stops at initializing). It works with FRSKY r-xsr running F-port, and seems to work with crossfire (stops at "downloading vtx tables" which I assume has to do with not running Betaflight on that quad)

All tested on BF 4.2
FRSKY R-XSR, fport - works fine
TBS Crossfire - Seems to be working
TBS Tracer - doesnt work, stuck at initializing

klutvott123, 2020-12-29:
Stuck on "Initialising" means telemetry isn't working properly. Double check wiring and configuration in BF, update to latest TX/RX firmwares and do "Discover new sensors" again in opentx.

The firmware version on your non BF quad has a MSP version that tells the scripts "This version of betaflight supports vtx tables" and that's why it's stuck at "Downloading vtx tables". I wouldn't recommend using the Betaflight lua scripts with other firmwares than betaflight even if it technically can work. If using these scripts with other firmware you may end up changing the wrong parameters on your FC.

oyvinla, 2020-12-29:
Thanks, stupid me forgot to turn on telemetry in betaflight. I was so sure I had it turned on, as I always do, but not on this quad obviously. 

For the non Betaflight quad, yes I'm aware, it was more for the sake of testing the link, not trying to actually use the lua script :)

so, it seems to be working fine with tracer as well

closed 2020-12-29 by oyvinla