betaflight/betaflight-tx-lua-scripts #379: Stuck on Downloading VTX Tables

ewingate, 2021-02-06:
I'm using a Jumper T16 with OpenTX (2.3.7), internal multi protocol module (
1.3.1.59), FrSky X4R with f.port (171114), and BF 4.2.2.

I have loaded all the folders for the 1.5.0 script onto my SD card and run the script to get it to compile. When I re-run it it initializes but then hangs on the Downloading VTX Table step. I was running the old version of the BF scripts and could change power and channel without any issue. I'm getting telemetry otherwise (RSSI, VFAS, etc). I've tried discovering new sensors.

Not sure what to do next.

klutvott123, 2021-02-06:
That it got past "Initialising" means that telemetry is working and the script is able to communicate with your flight controller. "Downloading VTX Tables" can take a while especially with Fport. Sometimes it can take a minute so just let it download for a while and see what happens. 

The X4R might be affected by a bug that prevents communication between the Betaflight lua scripts and the flight controller from working properly https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151.

ewingate, 2021-02-07:
Is there a workaround where I can load the vtx table manually?

klutvott123, 2021-02-07:
Yes, the vtx tab in the configurator has a button for that. It creates a .lua file containing the vtx tables. You have to rename the file to match the name of the model on your transmitter and then place it in the "BF/VTX" folder.

I recommend updating the receiver with the latest firmware version found here https://github.com/FrSkyRC/Firmware-Test/blob/master/ACCST%20firmwares%20v2.1.0/FW-X4R-XSR_ACCST_v2.1.0_FPort.zip as that might fix your issue. The multiprotocol module may also have to be updated for it to work with ACCST V2

ewingate, 2021-02-07:
I think the workaround is enough to close this. I put the table .lua file in the VTX folder and everything is working as intended now.

klutvott123, 2021-02-07:
Here's a simple test you can do to determine if it's REALLY working as intended. In the lua script try setting one of the pids to 125 and save the page. Did it work? Then try setting it to 126 and save the page. Does it still work?

ewingate, 2021-02-07:
I can save values up to 124. At 125 and above I don't get values in any of the PID fields but the value is set when I look in the Configurator. Setting the PID back to default in the Configurator and saving puts the correct values back into the script display.

klutvott123, 2021-02-07:
It's a bug in the receiver firmware and that's why the script couldn't download the vtx tables. The firmware I linked to above is the latest for the X4R and will probably fix the issue.

ewingate, 2021-02-07:
It has fixed the issue.

HirschJunge, 2021-02-23:
Hey, i have a DJi Quad and the lua script is stuck in downloading VTX table. Is there a workaround for this problem?

klutvott123, 2021-02-24:
Which receiver? Update RX/TX firmware.

HirschJunge, 2021-02-24:
R-XSR and RM TX16S Multimodule. Okay i will try that

closed 2021-04-03 by github-actions[bot]

peteoz, 2023-01-21:
I'm stuck on "Downloading board info", is this issue related ? 
Downloading VTX tables took a while but it seems eventually downloaded, the next screen was "downloading board info"

is there a work around whereby I could place the board info on the SD card manually ? Otherwise I cannot enter "Betaflight Setup" menu.

I'm running Taranis X9D+ with 2.3.15 OpenTX firmware