betaflight/betaflight-tx-lua-scripts #381: We can't open bl lua script after flashing 2.3.10 radiomaster tx16 ExpressLRS version

badRXx, 2021-03-10:
I came from a stock 2.3.10 opentx to the latest opentx for ExpressLRS, I used companion to flash the .bin and I copied the 2.3.10 sd contents. Then I added the expressLRS script, no problems with it, it opens and works fine, but when I try to execute the bl lua script can't be opened, I've tried with the 1.5.0 and previous versions, none of them can't be run, I only can see the message "initializing":

https://we.tl/t-tbNA4o9sX7

What's going on?

klutvott123, 2021-03-10:
Telemetry isn't working properly. Has telemetry been fully implemented in ExpressLRS?

lshems, 2021-03-10:
Opentx 2.3.10 has bugs on LUA. Use 2.3 11

badRXx, 2021-03-11:
Tahnk yo. I'm going to try 2.3.11, but I don't think I'm going to be able to flash the internal module for expressLRS, the 2.3.10 I used was an special version for ExpressLRS before the 2.4 is released (in a few years maybe) .

klutvott123, 2021-03-11:
@lshems I can't find anything about LUA bugs in the 2.3.11 release notes. Could you explain or share a link, please?

qdrk, 2021-03-13:
My betaflight lua script doesn't work for 2.3.11 either. I think full telemetry is just not implemented yet?

klutvott123, 2021-03-13:
From https://github.com/ExpressLRS/ExpressLRS:

> However, only basic telemetry is currently provided (VBAT, downlink/uplink LQ and downlink/uplink RSSI), work is underway for full telemetry support.