betaflight/betaflight-tx-lua-scripts #382: LUA changes of PID not saved to FC

jabacz, 2021-03-22:
I'm trying to set PID values in BF config scripts on OpenTX.
The number looks like changing, but nothing is propagated to actual FC setup.
Refresh of the PID screen re-reads previous values.
PID can be changed with BF configurator or using OSD config sticks commands.

Setup
F7 FC (MatekF722, FLAWOOF7DUAL)
R-XSR with S.Port telemetry (separate wire, no F.Port)
Taranis xLite OpenTX 2.3.11
Betaflight  4.2.x and 4.3 (several versions tried, no sucess)

klutvott123, 2021-03-22:
You need to save the page. Long press enter to bring up the popup menu. Select "save page" to send the modified values back to the flight controller.

jabacz, 2021-03-22:
Thanks. I'm feeling idiot now.

closed 2021-03-22 by jabacz

nicovv44, 2022-05-18:
@jabacz Not idiot at all, you helped others finding out as well!
Issues are sometimes like stack overflow, so I'm happy I landed on your question to find out the answer 😉

(People would say to just read the doc, but y'all must agree it's more convenient to just ask Google 😅)