betaflight/betaflight-tx-lua-scripts #387: Date and time not synced

TheIsotopes, 2021-04-18:
Testing bfbkgd.lua script on BF 4.3.0 latest master. Date and time was not synced on my MATEKH743. With BF 4.2.8 i have the same problem.
I'm using RadioMaster TX16S with OpenTX 2.3.11 the MM have v1.3.2.61, the RX is a FrSky R-XSR in Fport Mode with ACCST v2.1.0
The lua scipt is setting up as a special funktion in OpenTX.

Any suggestions?

klutvott123, 2021-04-18:
That's strange. Just checked my logs and all of them have the correct time and date. 4.3.0 with R-XSR FPORT 2.1.0, OpenTX 2.3.11 on QX7.

Fport telemetry is slow so it may take a few seconds for the date and time to be synced. Keep in mind that a special function only applies to the model it was created for while a global function works for all models. If you're running a one time script(such as the actual betaflight lua script) it will stop all other scripts to free up resources.
If you start logging before the script has successfully synced date and time I don't think itwill work.

TheIsotopes, 2021-04-18:
how can i see that the script is running and can i see the RSSi value in the configurator?

TheIsotopes, 2021-04-18:
The MatekH743 is the only Fc where i operate an SD card, so I only activated the script in this model. 

here you can see the contents of the sd card, all with the same date and time. 
[image]

klutvott123, 2021-04-18:
If it's set up like in the example in the readme and no one time scripts are running it should be running. Do you have any other scripts running that communicates with through MSP?

What's your ```blackbox_mode``` set to? if it's ```ALWAYS``` then the scripts might not have the chance to set the time before the file is created. 

With Fport you already have RSSI. The script detects if there's another rssi source and only sends rssi if there's not.

TheIsotopes, 2021-04-18:
everything is installed and set as described in the readme. other scripts are not executed.

`blackbox_mode` is set to `NORMAL` (default setting)

you are right, fport shows rssi without further ado. script deactivated and rssi is still displayed in the configurator.

klutvott123, 2021-04-18:
Try adding ```playTone(400, 100, 0, PLAY_NOW)``` inside this if statement. This will play a short beep when the time has been sent to the flight controller and a confirmation has been received.
https://github.com/betaflight/betaflight-tx-lua-scripts/blob/6fe2065ed09045e1618d87415162947271c71539/src/SCRIPTS/BF/data_init.lua#L15-L17

TheIsotopes, 2021-04-18:
don't get a beep to hear. 

[image]

klutvott123, 2021-04-18:
Try it here then 
https://github.com/betaflight/betaflight-tx-lua-scripts/blob/6fe2065ed09045e1618d87415162947271c71539/src/SCRIPTS/BF/data_init.lua#L21

If the script is running and doing something this should beep every 500ms

TheIsotopes, 2021-04-18:
unfortunately there is no beep to be heard.  :-((

klutvott123, 2021-04-18:
All right. Try launching "Betaflight setup" from the tools menu and exit it. This should make the background script restart.

You could try some beeps in bfbkgd itself. https://github.com/betaflight/betaflight-tx-lua-scripts/blob/master/src/SCRIPTS/FUNCTIONS/bfbkgd.lua. I suspect that the script isn't running at all or is exited because of some unknown error. Function scripts won't show an error if anything goes wrong. I doubt that there's anything wrong with the script itself because it works perfectly here and this is the first time I'm hearing about this.

TheIsotopes, 2021-04-18:
`bfbkgd.lua` script is executed. have inserted the sound in different places and it was played. 

I insert the line before data_init.lua is called then nothing happens.

klutvott123, 2021-04-18:
"Discover new sensors" again. "Tmp1" is needed for this to work.

TheIsotopes, 2021-04-19:
OMG, the bug is found. I have sorted the sensors and partly renamed them (e.g. Tmp1 -> Tmp).

Thank you for your help @klutvott123

closed 2021-04-19 by TheIsotopes

TheIsotopes, 2021-04-19:
@mikeller what is there to laugh about, in the readme there is nothing that the sensor tmp1 is explicitly required !? ;-))

mikeller, 2021-04-19:
@TheIsotopes: Nothing really. :-P Just that 'cleaning up' sometimes can have unintended consequences - which reminded me of those support calls from people who decided to clean up their hard drive by removing all of those unused files from `C:\Windows\`...

But you are right, the dependency on sensors, and the usefulness of running 'Detect sensors' when trying to get it to work, should probably be documented.

klutvott123, 2021-04-20:
Yes, this should have been better documented. There's no way for the user to know this. We should add a section about the telemetry requirements. I'll put it on the list of stuff to do before the next release. 😸