betaflight/betaflight-tx-lua-scripts #405: waiting for connection

izis1, 2021-11-10:
my latest working version.  v20210521.2833
newer stops on "waiting for connection"
qx7access, edgetx2.5.0, elrs 2.0.0-RC1 6f798a

klutvott123, 2021-11-10:
Is the flight controller and receiver powered on? Is telemetry working?

izis1, 2021-11-10:
Yes. quad with connected battery. I'm just change lua version on the radio.  v20210521.2833 OK. v20210521.2834 freeze waiting connection. am i doing something wrong?

klutvott123, 2021-11-10:
Can you test this, please? 
[betaflight-tx-lua-scripts_1.5.0.zip](https://github.com/betaflight/betaflight-tx-lua-scripts/files/7515950/betaflight-tx-lua-scripts_1.5.0.zip) 

For crossfire we're using the sensor "1RSS" to check for a connection, but express lrs may not have this sensor. The build above is baiscally v20210521.2834, but it uses the ```getRSSI()``` function check for a connection. This should work for all protocols.

izis1, 2021-11-10:
stops at Downloading board info

klutvott123, 2021-11-10:
Yes, it's a bug. You need Betaflight 4.2.11 or the latest master for it to work.

izis1, 2021-11-10:
Thanks. I will try tomorrow on the new version of bf4.3

izis1, 2021-11-11:
I tested the master compile, then the hex from pullrequest and then 4.2.11. Everyone stays on Downloading board info

klutvott123, 2021-11-11:
Thanks for testing. Looks like the fix didn't fix it then. I'll investigate

klutvott123, 2021-11-12:
@izis1 I have now tested it with tbs tracer and it works as it should. Without https://github.com/betaflight/betaflight/pull/11059 it hangs on "Downloading board info", but works as it should with the fix. 
It could be an issue with msp over telemetry in express lrs. 😕

closed 2021-11-30 by haslinghuis