betaflight/betaflight-tx-lua-scripts #414: Support for slider tuning

jasc76, 2021-12-21:
Will there be a version with slider support (BF4.3)

haslinghuis, 2021-12-21:
Please take a look: https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases/tag/v20211220.5042

ElRouge, 2022-01-10:
I tried but it gets stuck on "downloading board info" did I do something wrong?  with bf 4.2 and the "official" lua worked
[image]

klutvott123, 2022-01-10:
Which rx system are you using?

ElRouge, 2022-01-10:
> Which rx system are you using?

Elrs 868

klutvott123, 2022-01-10:
update to version 2.0.1

ElRouge, 2022-01-10:
No 1.2.1

klutvott123, 2022-01-10:
Then it won't work. at least 2.0.1 is required for it to work

ElRouge, 2022-01-10:
> Then it won't work. at least 2.0.1 is required for it to work

Ah ok i upgrade It ;-)

closed 2022-01-17 by klutvott123