betaflight/betaflight-tx-lua-scripts #416: Stuck at "Downloading board info" message on some boards.

ase-dcexpert, 2021-12-28:
I installed the BF 4.3.0 RC1 using Configurator 1.8.0 RC1 on two of my quads (pink and grey CineRace 20). Both quads use the same ELRS receiver but different FC boards (pink: CRAZYBEEF4FR, grey: GEPRCF411).

On my Taranis X-Lite S I installed the latest nightly build of the LUA script:
https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases/tag/v20211223.5115

The LUA script works fine with the pink quad and stuck at the grey quad during initialization with the message "Downloading board info".

I attached the "diff all" output from both quads. Maybe you can figure out why the grey quad is not initializing the LUA script and stuck with "Downloading board info".

klutvott123, 2021-12-28:
Looks like telemetry isn't enabled

ase-dcexpert, 2021-12-28:
UPDATE:
I have a third quad running BF 4.3.0 RC1 (URUAVHD85) with the same board (CRAZYBEEF4FR) like the pink (LUA working) quad.. this URUAV is also stuck at "downloading board info"

Telemetry is activated on all three quads!

CLI diff all attached
[BTFL_cli_URUAV_85HD_4.3.0.RC1_LUA_NOK.txt](https://github.com/betaflight/betaflight-tx-lua-scripts/files/7784449/BTFL_cli_URUAV_85HD_4.3.0.RC1_LUA_NOK.txt)
d

klutvott123, 2021-12-28:
are you using expresslrs 2.0.1?

ase-dcexpert, 2021-12-28:
Thats the difference.. only the pink CineRace had the 2.0.1 ELRS flashed.. the others had 2.0.0!

I just flashed ELRS 2.0.1 and now it is also working with the other quad! Thanks! I did not notice, that such a small difference causing the LUA to stuck!

closed 2021-12-28 by ase-dcexpert