betaflight/betaflight-tx-lua-scripts #418: Get stuck Initializing

akurekin, 2021-12-31:
Hi, I am trying to use betaflight lua script and it has got stuck at initializing stage. 
Installation: I copied v1.5 scripts to the SD card at Radiomaster TX16S, The first time I run the script it compiled successfully. The second time of running it got stuck with the message "Initializing" on the screen. Nothing seems to be wrong in my settings. I have updated the firmware on my TBS Micro transmitter and Nano receiver to the latest version and updated  Betaflight to 4.3 RC5 version. The "Telemetry" option is on, I can see the telemetry in my OSD and Radiomaster screen. I don't know what else I can check, but the LUA script does not want to work.

klutvott123, 2021-12-31:
First you should try this https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases/tag/v20211229.5241 for betaflight 4.3 support. Let us know if that doesn't work for you

akurekin, 2022-01-01:
Yes, the nightlies works for me,  Problem solved! Many thanks!

closed 2022-01-01 by akurekin