betaflight/betaflight-tx-lua-scripts #422: CMS functionality via Lua BF don't work - BF 4.3.0 RC2

Martivip, 2022-01-11:
CMS functionality via Lua BF don't work, on my Taranis X9D+ 2019 SE I only see blank screen.

After launch the latest BF nightly Lua Script (Betaflight CMS) I do not see any menus only the blank screen, under the upper right corner there is Refresh information with + sign but pusching any button don't refresh at all.
 
My configuration:
Taranis X9D+ 2019 SE with 2.3.14 OpenTX
Latest BF Nightly LUA script v20211229.5241
TX Namimo Flash OLED with 2.0.1 ELRS
Integrated RX in AIO Board, target: BETAFPVF4SX1280 with installed BF 4.3.0 RC2

[image]
[image]

klutvott123, 2022-01-11:
The cms script only works with TBS crossfire/tracer at the moment

Martivip, 2022-01-11:
@klutvott123 ok thanks for the information

closed 2022-01-11 by Martivip