betaflight/betaflight-tx-lua-scripts #433: Telemetry Protocol not Supported (TX16S, EdgeTx, Crossfire)

callagga, 2022-03-03:
If I run "Betaflight CMS" I get "telemetry protocol not support", but 
if I run "Betaflight Setup" I get the Betaflight config screen fine, get look at my rates and get the correct values, but when I change them (change value and push enter) it doesn't update them on my Betaflight.  (scripts did compile first time around)

Any ideas?

* Latest BF LUA scripts - v1.5
* Betaflight 4.2.9
* TX16S transmitter with:  EdgeTX v2.6.0.   Says OPTS: crossfire, ghost, afhds3, internalmulti, internalaccess, multimodule,luac
* Picking up telemetry fine on my TX16S from Betaflight...

klutvott123, 2022-03-03:
The CMS script isn't part of 1.5. Are you sure you're not using a nightly build? Anyway, Betaflight 4.3 is required for the CMS script.
For Betaflight setup you need to open the menu and select save. See the gif here https://github.com/betaflight/betaflight-tx-lua-scripts#betaflight-setup

callagga, 2022-03-03:
ok - I'll double check

callagga, 2022-03-03:
thanks this worked - I had initially grabbed the main branch and realised the mistake and cleaned up...but obviously didn't clean up properly

closed 2022-03-03 by callagga