betaflight/betaflight-tx-lua-scripts #434: compiling & testing doc (dev environment)

callagga, 2022-03-06:
Hi - Is there any additional information (links) that give some more background on how to modify/compile/test changes to these lua scripts?   In particular I would love to understand more regarding:

a) development env requirements & building
b) Is there a way you are able to test/run from Betaflight Companion and observe log info (print statements)? 
c) If no for (b) above do you have to deploy to your transmitter & run here to test?  If yes, then is there a way to debug or observe "print" outputs (log file)?  Wondering you manage tracking down bugs, testing to emulate etc 
d) any document that describes the interfaces for MSP calls (e.g. change Rates values) and the process (writing/reading)

klutvott123, 2022-03-06:
a) https://github.com/betaflight/betaflight-tx-lua-scripts#building-from-source or just do "./bin/build.sh" from the project folder. All this does is create the ```scripts.lua``` file which contains a list of all .lua files to be "compiled" on the first run of the script. 

b) Yes, the scripts can run on the simulator in the opentx companion and you can use ```print()``` to output information to the debug window. 

c) You can write directly to the lcd or to file using the io library. 

d) Not really, but here's the code https://github.com/betaflight/betaflight-tx-lua-scripts/tree/master/src/SCRIPTS/BF/MSP and here's the code for msp over telemetry in betaflight https://github.com/betaflight/betaflight/blob/master/src/main/telemetry/msp_shared.c

callagga, 2022-03-06:
@klutvott123 thanks very much - can I ask:

re a) I can run this on a MacBook in terminal then?  (I can try) as opposed to linux...

re b) if running/testing the scripts on Companion how would they work in terms of interacting with a Betaflight quad instance however?  e.g. if you wanted to test "changing rates" how would the betaflight lua scripts on Companion work here?    e.g. when I try to run "Betaflight Config" in Companion under Tools/"Betaflight Setup" it gets stuck on "Initialising" which I thought would make sense as it has no quad/betaflight instance to connect to?

klutvott123, 2022-03-06:
If it ca do make and .sh scripts it should work. 

It won't be able to communicate with the quad while running in the simulator. You need to manually set ```apiVersion```  and possibly some flags for it to get past the init stage without a connection. Have a look at https://github.com/betaflight/betaflight-tx-lua-scripts/blob/master/src/SCRIPTS/BF/ui_init.lua to see what's done at init. everything in that file relies on a connection to work.

closed 2022-04-13 by github-actions[bot]