betaflight/betaflight-tx-lua-scripts #437: Using last nightly BF lua script put the quad in a "telemtry lost" loop

LoakAudio, 2022-04-26:
Tested with OpenTX 2.3.14 and 2.3.15 with the same behavior : Before arming the quad for the first time (FC CrazybeeF4SX1280), the last nightly script (4 feb) seems to work fine. After arming, flying a bit, and disarming, launching the script create a loop of "telemetry lost" - "telemetry recovered" - "telemetry lost" ... without recovering full telemetry and the script can't update values. I need to remove the battery of the quad and shutdown the radio to recover telemetry.

Owning an other quad but with BF 4.2.11 wired to an EP1, the script works normally with this config.

My setup : Jumper T-Lite with the dedicated HappyModel ELRS module
                  OpenTX 2.3.14 and 2.3.15
                  HappyModel Crux 3 NLR  ( FC CrazyBeeF4SX1280 / spi ELRS Module)

Remaining at your disposal if you need more details.

klutvott123, 2022-04-27:
There's not much that can be done in the lua scripts to fix this. msp over telemetry is not working 100% as it should with spi elrs, and I would recommend using the configurator to configure your flight controller for now. Writing to flash doesn't work for spi elrs either when the command is sent using msp over telemetry.
We're aware of these issues and we have to find a solution for it. Thanks for reporting 👍

klutvott123, 2022-04-30:
I don't have the equipment to test that, but I believe it should be fine. Make some changes, save and exit. Did it work? Does the changes persist after a reboot? If so it should be fine

klutvott123, 2022-04-30:
@LoakAudio What framerate and telemetry ratio are you using?

LoakAudio, 2022-05-01:
Hi @klutvott123 ,

I've tested this morning with OSD and all I've tested (PID & Rate) work fine.

I'm at 50 Hz / 1:8 telemetry ratio.

I've rechecked the issue I've describe in my first post BUT it did'nt happened : no more "Telemetry lost" loop. Most of the time mods I've done (in Simplified PID & Rate, this is mainly what I want to modify in the field) show after a reload of the page of the LUA script but disappear  when I verify in the configurator. I think it's what's you are working on in #11573.

The main difference I seen is that I'm not on the field but on my desk with a fan cooling the FC. These FC AIOs tend to behave like barbecues.

I don't know what to say, except thank you for the time you gave me. I suggest that you leave this issue on hold, I will come back to it if I can correctly identify the conditions that trigger it.

I'm going to have a test of the firmware you linked in #11573

Thanks again

klutvott123, 2022-05-01:
> I'm at 50 Hz / 1:8 telemetry ratio.
This. The telemetry rate is extremely low even with 1:8 ratio and msp over telemetry will take priority over regular telemetry and this causes the radio to give the telemetry lost warning. I see this behaviour when using crossfire locked at 50Hz too. For better performance I would recommend at least 250Hz. Then the ratio can be dropped to 1:128 even. I find that higher framerate helps more than higher telemetry ratio. The issue could be that the buffers fill up faster than the RX is able to send telemetry. If there's no more room in the buffer, we end up with incomplete frames and the lua scripts will have to request the page again only to fail again and again.

It's not possible to save the changes you make with the lua scripts at the moment because eeprom write is disabled for elrs spi in the firmware itself. That's what I'm working on in #11573, yes

LoakAudio, 2022-05-01:
@klutvott123  I don't if I can interfere in the pull request #11573 as it could be reserved for devs so I do it here.

I've upgraded the FC with the firmware you linked for CrazyBeeF4SX1280 and so far so good. Rate and telemetry is still 50 Hz / 1:8. No loop, Telemetry is lost but only when the script updates the FC (I think) and regains fast. Mods in "Simplified tuning" and "Rates" stay when I verify in the configurator and a reboot by the LUA is just a reboot without any issue. More, values in the Lua (when I open the script) seem to be updated faster than before.

Just sometimes the quad beeps 2 times and it shows the visual beeper in the OSD (*****) 2 times too, I'm not really triggering something, just moving on a list of the LUA. I don't know what he's trying to tell me.

As I'm leaving in a flat, downtown, flying tests are not easy but tests on the deck are.

I hope this has been helpful.

klutvott123, 2022-05-01:
@LoakAudio Anyone is allowed to comment on PRs 🙂  Thanks for testing. 
Telemetry will be lost for a brief moment after writing to flash until the RX recovers again. Writing to flash is a special event that prevents anything from working normally until it's done.

closed 2022-12-23 by klutvott123