betaflight/betaflight-tx-lua-scripts #441: Betaflight Setup stuck on "initialising", Betaflight CMS shows "Syntax Error"

lahmann, 2022-06-16:
After updating to Betaflight 4.3.0, OpenTX 2.3.15, Lua-Scripts 1.6 and TBS CRSF MicroTX V6.17 the Betaflight Lua scripts are not working anymore. When accessing the "Betaflight Setup" Page on my X7S it is stuck on "initialising", also "Betaflight CMS" says "Syntax Error. Telemetry protocol not supported". I checked previous issues #340, #372, #433 with similar problems but could not resolve it.
Telemetry is activated in Betaflight and works as I can read the sensors in the radios telemetry screen. Rediscovering the sensors didn't help either.
Before the update I had a working setup with Betaflight 4.2.11, OpenTX 2.3.9 and MicroRX Ver 3.69 (iirc).

My setup:

Mamba F405 (FuryF4OSD) (Betaflight 4.3.0)
TBS CRSF NanoRX, MicroTX (both on 6.17)
Taranis QX7S (OpenTX 2.315, lua-scripts 1.6)

klutvott123, 2022-06-16:
Are you sure you're using version 1.6.0? that you're seeing "initialising" tells me you're not, as this message has been changed to "waiting for connection" in 1.6.0. where did you download the script from? 
I just installed 4.3.0 and 1.6.0 and it works just fine with tracer. my radio is a qx7 with 2.3.15

JhonyWill1108, 2022-06-17:
I use an FC FURYF4OSD with Betaflight 4.3.0, TBS CRSF nanoRX 6.17, Taranis X9D Plus with EdgeTX 2.7.0, I couldn't use Lua Script version 1.5.0, but version 1.6.0 worked perfectly

lahmann, 2022-06-17:
Yes, I did copy version 1.6.0 onto my SD card but thanks to your hint that you changed the message in 1.6.0 I found by grepping through the SCRIPTS folder that a compiled older version was still present on the SD card. This one was used instead of the new version. So now everything's working fine with Betaflight 4.3.0 - even Betaflight CMS!

Thanks for your time!

closed 2022-06-17 by lahmann