betaflight/betaflight-tx-lua-scripts #442: Stuck on "Downloading board info" with Crossfire

kristjanbjarni, 2022-06-17:
When using Crossfire and EdgeTX the script gets completely stuck on "Downloading board info".

Hardware/Software used:
- Crossfire Micro TX v6.13
- Radiomaster TX16S running EdgeTX 2.7.1
- Betaflight TX Lua Scripts 1.6.0

klutvott123, 2022-06-17:
Sounds like it's an issue with EdgeTX. Which BF version? 
I haven't had any issues downloading board info with opentx, crossfire, tracer, 1.6.0 and 4.3.0

kristjanbjarni, 2022-06-17:
BF version is 4.2.8.

I also tested on Jumper T16 running OpenTX 2.3.14 with same results so this doesn't seem to be EdgeTX exclusive.

I also did testing on some older nightly builds and found the same issue, but build v20211024.4306 seems to be one of the last versions to work, but I didn't test every version.

klutvott123, 2022-06-17:
you need 4.2.11. There was a bug https://github.com/betaflight/betaflight/pull/11059

beautown87, 2022-06-18:
This same thing occurs for me. I'm running Edge 2.7, Betaflight 4.3 with a mamba F405 MK2 flight controller. Note this seems to occur for select FCs as I have a different mamba F405 Mk3 DJI and the lua works with that.
@klutvott123  are you saying we need to roll back to 4.2.11 and the corresponding LUA if we want LUA functionality? Thanks

kristjanbjarni, 2022-06-18:
OK that seems to be the issue. I updated to 4.2.11 and it fixes the issue. I didn't realise that this version and higher was required since there weren't any BF version requirements listed. It would probably be a good idea to state this requirement for the release since a lot of people will probably run into this.

klutvott123, 2022-06-18:
@beautown87 Which rx systems are you using? Is it the same for all the flight controllers you've tried it with? If it's ELRS you will need at least 2.0.1 for it to work.
BF lua 1.6.0 is backwards compatible, but for BF version 4.2 you have to use the latest maintenance release which is 4.2.11. Best practice for the lua scripts is to use the latest stable release. 

@kristjanbjarni Done https://github.com/betaflight/betaflight-tx-lua-scripts/releases/tag/1.6.0

beautown87, 2022-06-19:
Hi @klutvott123  I'm using ELRS V2.3.0 and I have Betaflight 4.2.11 on my quad so that is probably the issue - thanks for the note above. I'm going to flash 4.3 on it and hopefully it will work with the new LUA. 
Note, I was not comparing apples with apples in my first note as quad 1 has BF 4.2.11 with mamba F405 MK2 (LUA 1.6.0 does not work) and quad 2 has has BF 4.3 RC 3 with mamba F405 DJI MK2 (LUA 1.6.0 does work).

beautown87, 2022-06-23:
Updating to 4.3 worked for anyone interested. Thanks for the advice @klutvott123

LeftStep, 2022-07-06:
I updated to 4.3 and am still having the same issue with LUA script 1.6...
Using EdgeTX 2.7.1 on TX16S.
FC is BetaFPV Toothpick F4 20A AIO V3 with BF 4.3.
Tx is XF micro and Rx is XF Nano both with Firmware 6.17.

Thomas72fpv, 2022-07-18:
Hello, same problem for me. I have BF 4.2.5 and the same message "downloading board info". After one hour, nothing happened...
Any idea ?

kristjanbjarni, 2022-07-18:
> Hello, same problem for me. I have BF 4.2.5 and the same message "downloading board info". After one hour, nothing happened... Any idea ?

As noted in the comments above you need to update BF to version 4.2.11 or higher since there was a bug fix that resolves this problem.

DaniLecx, 2022-07-24:
I'm also having the same issue, I'm stuck with "Downloading board info".
Here's my setup:
- EdgeTX 2.7.1 on QX7
- Betaflight 4.3.1
- Betaflight TX Lua Scripts 1.6.0
- Express LRS 2.0.0

haslinghuis, 2022-07-24:
@DaniLecx see comment above:
> If it's ELRS you will need at least 2.0.1 for it to work.

dlwlsns, 2022-07-25:
Having the same issue, scrpit build and VTX table both complete the process but it get stuck on "Downloading board info".
Setup:

- EdgeTX 2.7.1 on Taranis X9D+
- Betaflight 4.3.1
- BF LUA Scripts 1.6.0
- Crossfire 6.17

closed 2022-09-02 by github-actions[bot]

klutvott123, 2022-09-14:
https://github.com/betaflight/betaflight-tx-lua-scripts/pull/447 should fix this