betaflight/betaflight-tx-lua-scripts #443: Feature request: Support for Ghost telemetry

mavenius, 2022-06-22:
For as much detail as I can think of, I'm using a RadioMaster TX16S with EdgeTX 2.7.1 and the Ghost protocol (BF 4.3 stable release.)

When I open either Betaflight CMS or Betaflight setup (for script versions 1.6.0 each,) I am presented with "Telemetry protocol not supported!"

Looking at the readme for this repo, I see that Ghost isn't listed as a supported protocol; if that's incorrect and I'm doing something else wrong, then I'm sorry for the trouble :-) 

Thanks!

daleckystepan, 2022-06-22:
Hi man,
it is on the way:
- https://github.com/betaflight/betaflight-tx-lua-scripts/pull/419
- https://github.com/betaflight/betaflight/pull/11242
- https://github.com/EdgeTX/edgetx/pull/1377

and new Ghost FW.

mavenius, 2022-06-27:
@daleckystepan Excellent; thank you for the hard work!

larry415, 2022-06-28:
https://github.com/betaflight/betaflight-tx-lua-scripts/issues/443#

closed 2022-08-06 by github-actions[bot]