betaflight/betaflight-tx-lua-scripts #444: betaflight lua does not work on radiomaster tx16s with edge tx

davifelix, 2022-07-08:
Hello everybody.
i installed the latest version of edge tx on my radiomaster and the latest version of betaflight lua script, but i can't get it to work. I've been trying for days but nothing.
could you help me please?
a thousand thanks

klutvott123, 2022-07-08:
which edgetx version is the latest? 
Which version of edgetx did you use before the latest? did it work? 
How did you download and install the lua scripts? 
Which rx protocol are you using?
Which version of Betaflight are you using it with?
How exactly does it not work? Errors? does it hang? does it start at all?

davifelix, 2022-07-08:
edgetx version 2.7.1
old version used 2.6.0
it worked but I don't remember which version of lua.
I downloaded the .zip file and copied it to the radio sd card, overwriting when requested.
use of lrs express protocol with a frsky r9m or d16 module with 4in1 radio module.
I use betaflight 4.2.4
in practice it crashes at the start of the script.

i know i have an incompatible version of betaflight.

however, on this vehicle the bf 4..2.4 and express lrs crashes at the start.
on another drone with betaflight 4.2.11 and frsky protocol it does the same thing .......

klutvott123, 2022-07-08:
Does it successfully go through the compile when executed for the first time?
Does it give you a error message?

I would try formatting the sd card and install the scripts again. backup the things you need.

davifelix, 2022-07-08:
ok I already tried, but tomorrow I'll do it again. I'll update you if I can get it to work. will it be necessary to install the previous version of the lua so as to be compatible with the 4.2.4 version of bf?

thanks for the help!

klutvott123, 2022-07-08:
Yes, you would need 1.5.0 for 4.2.4(or any 4.2 version pre 4.2.11). I recommend just updating to 4.2.11 instead as there are no major changes, only bugfixes, so you should be able to use the same settings without issues.

davifelix, 2022-07-10:
everything OK! in the end it was the "wrong" version along with a sd full of errors.
thank you very much for helping!

closed 2022-08-10 by haslinghuis