betaflight/betaflight-tx-lua-scripts #445: bdbkgd lua reliable only once per ExpressLRS TX session

pitts-mo, 2022-08-06:
Because I rarely experienced this issue when using TBS Crossfire I initially thought this issue to be specific to ExpressLRS. See https://github.com/ExpressLRS/ExpressLRS/issues/1616 However, further testing where I attempt to trigger multiple instances of bdbkgd lua also seem to work once per ExpressLRS session. This suggest to me that the bdbkgd lua is hanging up or becoming stuck for extended periods while ExpressLRS is being utilized.

There do not appear to be many people able to assist with this issue :-/ Can someone help me isolate this problem so that either the bdbkgd lua can be updated or a specific issue can be raised for the ExpressLRS project?

1. Describe your problem; Each instance of bdbkgd lua is able to push RTC (Real Time Clock) to FC reliably once per ExpressLRS TX session. Rebooting the TX or triggering a new instance of bdbkgd lua will again allow RTC to FC to occur reliably only once during the ExpressLRS TX session.
2. Include ways to reproduce the problem; from all recent builds of OpenTX or EdgeTX set bdbkgd lua to trigger on tel and power up a quad with Betaflight FC osd_rtc_date_time_pos element displayed within OSD and without an alternate time source configured (no GPS).
3. Provide as much information as possible about your hardware and software, including:
-OpenTX or EdgeTX (all builds over last year)
-ExpressLRS builds from 2.0.0 to 3.0.0-rc1
-observed to occur on all Betaflight FC and configurations I have tested. Note that all similar configurations tested with TBS Crossfire utilized were reliably reporting RTC in OSD.

pitts-mo, 2022-09-21:
This is still observed to occur on ELRS 3.0.0 final release (https://github.com/ExpressLRS/ExpressLRS/commit/8aa1b03e804ee896e974a05c3d9aadfb93d17069).
Any suggestions as to where I can look to try and narrow down the cause of this?
-p

klutvott123, 2022-09-21:
the background script uses the getRSSI function to detect a connection. When your quad is powered off, the function should return 0 as there is no connection, and the script should "reset" so that it can do its thing again when a connection is detected.
https://github.com/betaflight/betaflight-tx-lua-scripts/blob/master/src/SCRIPTS/BF/background.lua

and https://github.com/EdgeTX/edgetx/blob/155a854131d142a1740dd3a57b40a94e98dcad6a/radio/src/lua/api_general.cpp#L1748-L1769

I would start by verifying that getRSSI is actually returning 0 when there is no connection.

pitts-mo, 2022-09-24:
Thank you @klutvott123  for this great starting point.

Using a modified lua I find getRSSI() displays 99 while ELRS is connected and -- after ELRS has disconnected (nil?).

using this info I have tried a few different script mods deleting the luac each time... but even the most basic hack attempt of updating background.lua line 7 from:
    if getRSSI() > 0 then
to:
    if getRSSI() == 99 then

still gets the same result where each lua instance is only able to insert RTC a single time into my betaflight OSD.

I will try to dig at this again later today but welcome any additional hits you can offer.
Thank you, -p

klutvott123, 2022-09-24:
I just did some tests with getRSSI, and it reports 99 when connected and 0 when not, as expected. The background script is very simple and I don't really see what could go wrong. I see that you have the script set to trigger on telemetry. Try having it always run when the radio is ON. It doesn't need to be manually reset by only having it run when telemetry is detected.

pitts-mo, 2022-09-24:
Thank you @klutvott123. Yes, changing the special function switch from Tele to ON does appear to have made my RTC reliable again. I will test some more and come back to close this. -p

klutvott123, 2022-09-25:
@pitts-mo When using telemetry to trigger the script it probably just stops it from running without terminating it. It's still loaded and it's state is preserved. Then when telemetry comes back it continues where it left off. For the script itself it would be like you never lost the connection and it wont reset itself so that the RTC is set again.

pitts-mo, 2022-09-25:
Makes sense. Thank you @klutvott123.

The solution was user re-education of suggested bdbkgd use.
eg:
Set bdbkgd switch to "ON" as shown in the readme https://github.com/betaflight/betaflight-tx-lua-scripts ... either as global function or within the model's special functions.

Sorry, I had started using Tele to trigger background scripts maybe 5 years ago but don't remember who gave me that idea or what project it was for.

closed 2022-09-25 by pitts-mo