betaflight/betaflight-tx-lua-scripts #448: Try to add new Page

druckgott, 2022-09-17:
I used newest version and try to add my own page for OSD. What I did:
I generate a new Page:
`local template = assert(loadScript(radio.template))()
local margin = template.margin
local indent = template.indent
local lineSpacing = template.lineSpacing
local tableSpacing = template.tableSpacing
local sp = template.listSpacing.field
local yMinLim = radio.yMinLimit
local x = margin
local y = yMinLim - lineSpacing
local inc = { x = function(val) x = x + val return x end, y = function(val) y = y + val return y end }
local labels = {}
local fields = {}

if apiVersion >= 1.041 then
	fields[#fields + 1] = { t = "vcd_video_system",     x = x + indent, y = inc.y(lineSpacing), sp = x + sp, min = 0, max = 2, vals = { 2 }, table = { [0] = "AUTO", "PAL", "NTSC" } }
	fields[#fields + 1] = { t = "osd_units",        	x = x + indent, y = inc.y(lineSpacing), sp = x + sp, min = 0, max = 2, vals = { 3 }, table = { [0] = "IMPERIAL", "METRIC", "BRITISH" } }
	fields[#fields + 1] = { t = "rssi_alarm",        	x = x + indent, y = inc.y(lineSpacing), sp = x + sp, min = 0, max = 100, vals = { 4 } }
end

return {
    read        = 84, -- MSP_OSD_CONFIG
    write       = 85, -- MSP_SET_OSD_CONFIG
    title       = "OSD Test",
    reboot      = false,
    eepromWrite = true,
    minBytes    = 3,
    labels      = labels,
    fields      = fields,
}
`

Add this to the page Menü.
If I call the page I get all 3 fields with the right Information.
But If I save this page It didn´t update the value on the Flightcontroller.

So I can read the things, but cant Update it.
Can anybody give me an advice what I´m doing wrong or how I have to change it to get It updated also.

BR
druckgott

klutvott123, 2022-09-27:
The first byte is not the same thing for MSP_OSD_CONFIG and MSP_SET_OSD_CONFIG. For MSP_OSD_CONFIG the first byte is osd flags, while MSP_SET_OSD_CONFIG expects an "address" as the first byte. Check the betaflight code for this
https://github.com/betaflight/betaflight/blob/a90e2e5fb1dfd27c7ab20fefbf11c808c9f6699c/src/main/msp/msp.c#L940
https://github.com/betaflight/betaflight/blob/a90e2e5fb1dfd27c7ab20fefbf11c808c9f6699c/src/main/msp/msp.c#L3961

What you need is likely a `preSave` function that modifies the first byte.

druckgott, 2022-10-01:
Perfekt thx I get it
```

local template = assert(loadScript(radio.template))()
local margin = template.margin
local indent = template.indent
local lineSpacing = template.lineSpacing
local tableSpacing = template.tableSpacing
local sp = template.listSpacing.field
local yMinLim = radio.yMinLimit
local x = margin
local y = yMinLim - lineSpacing
local inc = { x = function(val) x = x + val return x end, y = function(val) y = y + val return y end }
local labels = {}
local fields = {}

if apiVersion >= 1.041 then
	fields[#fields + 1] = { t = "adre",                 x = x, y = inc.y(lineSpacing), sp = x + sp, min = -1000, max = 10000, vals = { 1 } }
	fields[#fields + 1] = { t = "vcd_video_system",     x = x, y = inc.y(lineSpacing), sp = x + sp, min = 0, max = 2, vals = { 2 }, table = { [0] = "AUTO", "PAL", "NTSC" } }
	fields[#fields + 1] = { t = "osd_units",        	x = x, y = inc.y(lineSpacing), sp = x + sp, min = 0, max = 2, vals = { 3 }, table = { [0] = "IMPERIAL", "METRIC", "BRITISH" } }
	fields[#fields + 1] = { t = "rssi_alarm",        	x = x, y = inc.y(lineSpacing), sp = x + sp, min = 0, max = 100, vals = { 4 } }
	fields[#fields + 1] = { t = "cap_alarm",        	x = x, y = inc.y(lineSpacing), sp = x + sp, min = 0, max = 20000, vals = { 5, 6 } }
end

return {
    read        = 84, -- MSP_OSD_CONFIG
    write       = 85, -- MSP_SET_OSD_CONFIG
    title       = "OSD Test",
    reboot      = false,
    eepromWrite = true,
    minBytes    = 5,
    labels      = labels,
    fields      = fields,
	preSave = function(self)
		self.values[1] = -1
        return self.values
    end,
	
}
```

closed 2022-10-01 by druckgott