betaflight/betaflight-tx-lua-scripts #450: Download vtx-table stuck and overheats vtx

0xphk, 2022-09-26:
I'm using Edgetx 2.7.1 on Radiomaster zorro with internal CC2500 multi module (1.3.3.7) 
and Betaflight 4.2.11 on an UR85HD with SPI FrSky receiver.

I can't get further than the 'downloading vtx table' step. 

As mentioned in #379 I did let it sit for a while but after 2mins I noticed a light smelling (plastics/electronics sort of.)
I removed power and noticed the vtx was heated up very badly.

I tried a second time after cooldown and measured the temperature. 
Normally when sitting powered up on the bench, it's on 25mW and about 66°C but as soon as it switches to the download vtx table step, 
it heats up to 76°C in about 10-15sec and starts smelling. You can't even touch the camera screws..

I don't want to risk burning the vtx, do you have any idea why it behaves so?
I tried creating the vtx.lua in betaflight and named it exactly as the model name, 
but it still gets to the download step and overheats and I have to cut power. 
The script then switches to 'waiting for connection'.

klutvott123, 2022-09-26:
Sounds strange. The bf lua scripts aren't changing the vtx settings at this point. It's just requesting the vtx table over and over until it succeeds. It can take a while, especially with frsky. I don't have any experience with spi frsky, but the situation is probably worse.

If you're using version 1.6.0 you can't use the model name for the vtx tables. In this version we're using the mcu id name the file so that each craft can have it's own vtx tables. You can generate the vtx tables file in the vtx tab of the configurator and rename it with the mcu id. you can find the mcu id in cli with the `mcu_id` command. Then place the file in `SCRIPTS/BF/VTX_TABLES`

0xphk, 2022-09-27:
Using the ```mcu_id``` as filename worked and I was able to see the menu after a few tries. First it was stuck at 'downloading device settings/information' (can't remember the exact phrase). Now it enters the menu after ```device id``` step. 

But there are no values for either VTX settings or rates / pids profiles or all the other settings. All just display ```---``` as value.
So I guess I'm out of luck for SPI based FrSky receivers?

But I can confirm, that using the vtx table lua no longer leads to overheating, now I'm back at ~66°C while sitting powered on thed desk.

klutvott123, 2022-09-27:
@0xphk "downloading board info" is done when the scripts detect a new flight controller, and only happens one time. "device id" is actually the mcu id which is used to identify the flight controller and has to be done each time the script is launched. The fact that you are getting past this stage tells us that communication is working and the scripts are able to retrieve the requested information from the flight controller. Not sure why the pages aren't loading, but there has always been issues with spi frsky.

0xphk, 2022-09-29:
I guess it's a problem with SPI based receivers. Tested a few more and never got it working. Some didn't even get past the 'board information' step. Already planning to replace most of them with ELRS receivers and then I will check the script again.

closed 2022-09-29 by 0xphk