betaflight/betaflight-tx-lua-scripts #455: Feature Request: blackbox managment

porlock, 2022-10-19:
It will be very handful if will be possible to check black box occupied/free space and purge the data.

haslinghuis, 2022-10-19:
Do you mean some OSD element or LUA warning?

porlock, 2022-11-27:
> Do you mean some OSD element or LUA warning?

Yes, same like you can change PID settings

porlock, 2023-04-25:
Just a menu for BBL management, that will allow to display how much space on flash memory left free / how much is already occupied , and also initiate flash erase. Similar like in BF Configurator

neprsitady, 2024-06-13:
@porlock you can erase the blackbox via the switch set in BF modes. Programm that to some leftover button or new radio model with different switches/channels for the "management/tune/failsafe" testing.