betaflight/betaflight-tx-lua-scripts #467: Stuck on "Downloading Board Info" page

Bradw-17, 2023-01-17:
Trying to update all my gear after not flying for a year or so and I'm having trouble with running the Betaflight LUA script.

After updating all the SD card contents and running the script, it compiles, then downloads VTX tables, but gets hung up on "Downloading board info" page. I waited around 10 mins or so but nothing changed. 

-Taranis QX7 - EdgeTX v2.8.0
-Crossfire Micro TX - v6.19
-Matek F405-CTR - betaflight_4.4.0_STM32F405-2933-56373817f.hex
-BF LUA Script - Tried both the latest official release 1.6.0 as well as nightly build v20221220.2

Any help would be very much appreciated!

klutvott123, 2023-01-17:
What's the frame rate?

Bradw-17, 2023-01-17:
@klutvott123 Sorry, I'm a little rusty at this. Could you clarify what you mean by "frame rate"? I know my RC link rate is locked to 150hz if you mean that?

klutvott123, 2023-01-17:
Yes, that's what I mean, thanks. I'll test this myself later to see if I can reproduce it.

klutvott123, 2023-01-17:
Works here. crossfire micro tx 6.19 150Hz, aikonf7 with latest bf master flashed from configurator, latest bf lua nightly, taranis qx7 running opentx.

Can you try going back to 4.3?

Bradw-17, 2023-01-19:
Sorry it's taken a bit for me to get back to this. 

I just flashed back to BF 4.3 and I seem to be having the same issue. Have there been any significant changes to radio configuration settings or anything that I may have missed?

klutvott123, 2023-01-19:
Good, then we can rule out 4.4 being the issue at least. Not that much has happened since last year. That it succeeds in downloading api version and vtx tables indicates that communication is working. The only thing I can think of that is "special" about the board info is that the message is large enough to not fit into one single crsf frame, but that hasn't been an issue with crossfire or tracer. 

I would update everything again just to be sure, and double check the tx and rx settings.

Bradw-17, 2023-01-19:
Checked my TX link and all my inputs are going through as well as receiving telemetry back from the RX. Just reflashed to 4.4 and downgraded the LUA script to v1.5, and it gets stuck at the "Initializing" screen (probably getting stuck at the same point). 

Really stumped on this lol

klutvott123, 2023-01-19:
Which crossfire firmware version did you upgrade from? There was a point where it needed to be updated twice for everything to work as it should.

Bradw-17, 2023-01-19:
Could have been anywhere from v6.10 to v6.16, I can't quite remember. 

I just flashed v6.19 again but I'm getting the exact same thing unfortunately.

klutvott123, 2023-01-20:
[betaflight-tx-lua-scripts_1.6.0.zip](https://github.com/betaflight/betaflight-tx-lua-scripts/files/10470077/betaflight-tx-lua-scripts_1.6.0.zip)
can you try this, please? 
It won't fix it, but it will log the replies in "bf_lua_log.txt" on the root of your sd card. 
From the tools menu, select "BF log msp" and let it try downloading board info for a few seconds, then exit the script. Upload the log file.

Bradw-17, 2023-01-21:
[bf_lua_log.txt](https://github.com/betaflight/betaflight-tx-lua-scripts/files/10471165/bf_lua_log.txt)

klutvott123, 2023-01-21:
well, that was unexpected. the last five entries in the log looks like the second frame of the response to MSP_BOARD_INFO. So where is the first frame, and why are you the only one experiencing this? 😄 
Could you try flashing back to the previous edgetx version?

Bradw-17, 2023-01-21:
Here's the log file after flashing back to EdgeTX v2.7.1
[bf_lua_log.txt](https://github.com/betaflight/betaflight-tx-lua-scripts/files/10471295/bf_lua_log.txt)

klutvott123, 2023-01-27:
Thanks. I still can't reproduce this with my crossfire micro tx v2 and nano rx. I have it connected to a ff fortini f4 right now. 
Which receiver are you using? Has this ever worked for you with the hardware you're using now? try doing a factory reset.

Bradw-17, 2023-01-27:
I'm using a Crossfire Nano RX. And yeah I've never had an issue with LUA scripts until now. If it's my only option I guess I'll try a factory reset.

ggoraa, 2023-01-30:
Well, I have the same issue here, also stuck on "Downloading board info", also latest Betaflight and EdgeTX(the same thing was on OpenTX, so EdgeTX should not be the culprit)

klutvott123, 2023-01-30:
@ggoraa Are you using crossfire too?

ggoraa, 2023-01-31:
Yep, I do, also Crossfire, latest FW version, Micro TX and Nano RX

klutvott123, 2023-01-31:
Which FC are you using? Has it ever worked for you before on the same hardware?

ggoraa, 2023-01-31:
FOXEERF22V2, and no, it never did

lionwebeu, 2023-02-03:
I have the same problem after update to EdgeTX 2.8.0 and BF 4.4.0 - the Betaflight Config Lua stop in "Downloading boar info" msg. My RC radio is TARANIS X9D PLUS 2019 with Crossfire Micro TX V2 (Firmware 6.13). FC IFLIGHT_F722_TWING.

Before all works good with BF 4.3.2, OpenTX 2.3.11 with previous version BF LUA 1.6.

What can I try to do to solve this problem?

lionwebeu, 2023-02-04:
[image]

I have good news... after change Boudrate from 115k to 400k - the BF LUA 1.7 works good.

The question is that is correct settings  "400k" or maybe some day fail when I fly? 
Becouse I read the default settings is 115k for my radio :)

klutvott123, 2023-02-04:
400k is what you want for crossfire, and it being at 115k would explain why it wasn't working as it should.  Everyone else please check your baudrate settings 🙂

Bradw-17, 2023-02-04:
Well what do ya know! Haha I never would have thought to check that but looks like it got changed after a radio update. 

Thanks a ton for the fix! Working perfectly now!

closed 2023-03-15 by github-actions[bot]

MrThing, 2024-05-25:
This should be added to the Wiki under Installation or Troubleshooting Issues. My radio was set to 115k and the 400k baud rate instantly changed it. 

Please add it.

pfeerick, 2024-05-25:
Why not do a PR and add it yourself?

btw, higher values than 400k should work with both Crossfire/Tracer as
newer firmwares sync automatically, and ELRS will just work with whatever
the fastest your transmitter hardware can reliably operate at. 115k with
crossfire/tracer is a controls only - no telemetry IIRC - (legacy?) mode
hence why it doesn't work... they recommend 400k as the minimum in the
manual.