betaflight/betaflight-tx-lua-scripts #469: Problema su Radio BetaFpv Lite 3 PRO

SimonTinyFPV, 2023-02-01:
copio i file nella directory come spiegato installo il BF Lua ma poi non riesco ad impostare la visualizzazione Script perche mi dice che non esiste il file nella SD

klutvott123, 2023-02-01:
english please. are you launching the script from the tools menu? which file does not exist?

SimonTinyFPV, 2023-02-01:
copy the file in the directory as explained install the BF Lua without risk to set the display Script because I say that the file does not exist in the SD

SimonTinyFPV, 2023-02-01:
https://user-images.githubusercontent.com/124176978/216118103-6a607088-977c-4b69-b2b1-2e3e7db966d9.mp4

[image]
[image]
[image]
[image]

SimonTinyFPV, 2023-02-01:
when I am in the Display category that I have to enable the LUA to see the PIDs on the radio display it tells me that the script is not present in the SD this problem does it with all versions on this radio

klutvott123, 2023-02-01:
Looks like you did everything right. to start and use the script, you have to select "Betaflight setup" from the TOOLS menu. There is no telemetry script anymore, so you don't have to do anything in the DISPLAY page 🙂

SimonTinyFPV, 2023-02-01:
ok thanks  I'll let you know tomorrow

SimonTinyFPV, 2023-02-02:
[image]

SimonTinyFPV, 2023-02-02:
No work 
[image]

klutvott123, 2023-02-02:
Is telemetry enabled in betaflight?

SimonTinyFPV, 2023-02-02:
Ok NOW work thanks @klutvott123 💪
[image]

klutvott123, 2023-02-02:
Awesome! 😁

closed 2023-02-02 by klutvott123