betaflight/betaflight-tx-lua-scripts #470: Suck on "Waiting for

undefine-man, 2023-02-01:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Remove this Text :).

[devices] : 
Jumper T16
edgeTX 2.8.0
Betaflight firmware : 4.3.0

[Betaflight setup] :
Suck on "Wait for API version"  when using "Betaflight setup" 
[image]

[Betaflight CMS] :
[image]

klutvott123, 2023-02-01:
That means there's no communication. Which rx/tx are you using? 
Also keep in mind that the FC and RX has to be powered on for it to work.

undefine-man, 2023-02-01:
i am using TBS crossfire nano rx and TBS crossfire Micro TX .  TBS lua works fine on my remote control.

klutvott123, 2023-02-01:
and do you have telemetry enabled in betaflight?

KAJE1977, 2023-02-05:
I have the same thing using ghost  newest firmware 1.8.0. Using betaflight 4.4.

klutvott123, 2023-02-05:
@KAJE1977 For the RX/TX, are you using a mode that supports telemetry? I use race250.
I'm not 100% sure if msp support is in any of the stable firmware releases for ghost, so you may need to try the latest beta

druckgott, 2023-02-05:
I have also a Problem with the script and also telemetry. I have frsky.
Protokoll FPORT and Telemetry FRSKY_HUB for BF4.4

klutvott123, 2023-02-05:
Not familiar with frsky hub, but looks like it doesn't support msp over telemetry.

druckgott, 2023-02-05:
On BF4.3 it worked all the way. Do I maybe need to select something differnet?

klutvott123, 2023-02-05:
In the configurator firmware flasher, try adding `TELEMETRY_SMARTPORT` under custom defines.

druckgott, 2023-02-05:
That was the solution thx.

KAJE1977, 2023-02-05:
I have it working now. I put TELEMETRY_GHST in the user defines and reflashed. Maybe it was that or i could have missed something.....

KAJE1977, 2023-02-05:
FYI i was also using Race mode which seems to have issues. It will still connect but some of the pages will not load. Other telemetry modes seem to be working fine.

klutvott123, 2023-02-05:
Yeah, some of the modes don't work too well for this. Race250 is the way to go for msp over telemetry performance.

klutvott123, 2023-02-05:
@druckgott http://betaflight.com/docs/development/API/Cloud-Build-API

I assumed you already had selected frsky hub as the telemetry protocol in the dropdown, and you can only select one protocol so the other one has to go into custom defines 🙂 

This will include smartport automatically when you select fport  https://github.com/betaflight/betaflight/pull/12314

druckgott, 2023-02-06:
perfekt thx

OCDFPV, 2023-02-11:
@klutvott123 
I am on the latest version of EdgeTX, Betaflight 4.4, configurator 10.9
I am using CRSF.  On all of my quads with Kiss FC's and running KISS firmware I have no issues. I am able to see the flight controller on the main page of the LUA Agent Lite. I can control the PID'S and see all relative FC info, but on all of my quads that use the Betaflight  firmware I can't see my flight controller on the CRSF Lua Agent Lite.
I also tried to use Betaflight Lua (latest version) on those quads, but it doesn't get past the initializing screen. It did initially go through the " compiling phase, but it never loads the FC info.
Any idea what my issue could be?
Thank you in advance!

klutvott123, 2023-02-11:
"initialising" indicates that you're using an older version of the betaflight lua scripts. Make sure you're using version 1.7.0.
Also make sure that telemetry is enabled on the "receiver" tab in the configurator.

OCDFPV, 2023-02-11:
Thanks for answering my question.
I am using betaflight lua scripts 1.7.0.
The Telemetry is enabled in the receiver tab.
It's only seems to get stuck on " initializing" on quads with CRSF. On quads with the Frsky protocol (RSXR) it works no problem.

klutvott123, 2023-02-11:
the "initializing" message was replaced a long time ago and is not present in 1.6.0 or 1.7.0. Install 1.7.0 again and let it overwrite the already existing files on your sd card when copying the new version over.

OCDFPV, 2023-02-11:
[image]
This is the screen it gets stuck on

klutvott123, 2023-02-11:
In edgetx you need to set the baud rate to 400k. see https://github.com/betaflight/betaflight-tx-lua-scripts/issues/467
🙂

OCDFPV, 2023-02-12:
I will change it to 400k now and message back

OCDFPV, 2023-02-12:
That did it!
Thanks @klutvott123 !

klutvott123, 2023-02-12:
Nice! Enjoy the BF lua scripts 👍

haslinghuis, 2023-02-12:
@klutvott123 do see a change adding this to documentation to not repeat our self.

Maybe we can add some timeout to the script including a message check documentation for required settings?

klutvott123, 2023-02-12:
@haslinghuis Yes, we could add the baud rate to the requirements. I'll look into adding some more info to the init stage 🙂

TTGG4400, 2023-02-15:
Is there a custom define that fixes a CRSF setup, like that of undefine-man, the OP?  I have that same setup almost - TBS CRSF Rx, TX16S, ETX2.8.0, BF4.4.0 and BF LUA v1.7 - and I get exactly the errors he posted, when I try to run the BF LUA script (which use to work, a few versions back).

klutvott123, 2023-02-15:
@TTGG4400 No, all you have to do is select CRSF as the RX protocol  in the firmware flasher tab. It depends on CRSF telemetry so that will be included automatically. make sure telemetry is enabled in the receiver tab, and that the baud rate is set to 400k in edgetx.

TTGG4400, 2023-02-15:
Thx - I've been doing all of that, except that my baud rate is set to Mode CRSF, 400k, but the Status says "150 Hz 1 Err".  WHOOOOHOOOO - I just saw that telemetry on my BFC Rx tab was disabled.  However, that was enabled at one time according to my saved CLI.  Dunno how it got disabled - Happened once before too.  I need to watch what I'm doing.  Thx for your help, BTW.

klutvott123, 2023-02-15:
@TTGG4400  It's a configurator bug that disables telemetry when changes are made in the ports tab. It will be fixed in the next release

TTGG4400, 2023-02-15:
> @TTGG4400 It's a configurator bug that disables telemetry when changes are made in the ports tab. It will be fixed in the next release

Wow - thanks a lot for that - I didn't think it was me, but, I wasn't sure.  You just removed a lot of self doubt from my daily life. THANKS.

RipperDrone, 2023-03-05:
@klutvott123 With a quad with Walksnail digital video on board I am getting to the script initialization point of 'Downloading VTX table', then the script gets stuck (presumably since there is no VTX table nor settings defined when using MSP for digital video). How can I get the v1.7 BF LUA script to work anyways?

klutvott123, 2023-03-05:
@RipperDrone The quick solution for now is to keep `VTX` selected in the cloud build options. You don't have to set up vtx tables or anything.

RipperDrone, 2023-03-05:
@klutvott123 Oh, nice one. Didn't think of that. Tried to enforce definition of a VTX table by enabling TBS SA, but no luck getting vtx tables defined unless the actual hardware is responding, it seems. Will try - thx a lot.

klutvott123, 2023-03-21:
@RipperDrone Try the latest nightly. It should work without having VTX defined.

closed 2023-04-29 by github-actions[bot]