betaflight/betaflight-tx-lua-scripts #476: Lua script hang on waiting for vtx

Pancronos, 2023-04-18:
I upgraded to 4.4.1 firmware on a diatone F740 AIO and removed led strip and vtx (have a caddx vista) in other options, when launching betaflight setup on RM-TX16s (latest lua downloaded yesterday) it hangs on vtx table. I thought it was going to timeout after the check for vtx and get all the menu, but it hangs on vtx table even after 30 seconds.
I saw a commit [Create empty vtx table if not available](https://github.com/betaflight/betaflight-tx-lua-scripts/commit/61d302980967f05f3e8a4957dfc45c859559cdf0) but this doesn't seem to work.
Flashed the firmware with vtx table and LUA script works fine.

Pancronos, 2023-04-18:
saw previous issue 
https://github.com/betaflight/betaflight-tx-lua-scripts/issues/470#top

closed 2023-04-18 by Pancronos

klutvott123, 2023-04-19:
try a nightly build https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases