betaflight/betaflight-tx-lua-scripts #477: Waiting for Api

druckgott, 2023-04-29:
#System
Taranis X-Lite Pro
Expresslrs 3.2.1
Betaflight 4.4.1
Current Version of this Script

## If you believe there is an issue with the firmware itself please follow these steps:
I have "Waiting for Api" if I go to the script
I have Expresslrs 3.2.1 with 400 baud rate
I have installed Smartport 

Hopfully I did not lost anything

klutvott123, 2023-04-29:
Make sure telemetry is still enabled in the receiver tab

druckgott, 2023-04-29:
Yes it's enabled I checkt it several times

haslinghuis, 2023-04-29:
Expresslrs does not provide smartport it uses the crsf protocol

druckgott, 2023-04-29:
Ok this I did I ly reinstall after I'd did not work with crsf Protokoll. But what could be case this issue? Can I see any log?

klutvott123, 2023-04-29:
is rx and telemetry working? the issue is that the script isn't getting a response to the request.

druckgott, 2023-04-29:
Yes that's the Stange point telemetry is working an rx is also working

haslinghuis, 2023-04-29:
Just installed nightly build on MERCURYG4 with ELRS 3.2.1 and `Betaflight setup` script seems to be fine.
But using `Betaflight CMS` I get only the main screen and no movement.

I have no VTX installed.
@druckgott what VTX setup do you have?

druckgott, 2023-04-30:
Betaflight I have compiled it, but I have nothing inside:
[image]

Is there a way to deaktivate it temporary (comment out) to check if this is the problem

I have DJI VTX

druckgott, 2023-04-30:
Ko I sed Baudrate to 3.75M and now it works ;-)

klutvott123, 2023-04-30:
lol that's strange. It has always worked for me at 400k. you have to use bf master if you want betaflight cms to work

druckgott, 2023-04-30:
Ah ok no it's not so Important, I only test it because you mentioned it

closed 2023-06-07 by github-actions[bot]