betaflight/betaflight-tx-lua-scripts #478: Fix for ''stuck on downloading VTX tables''

Hillbilly-Phil, 2023-05-05:
If you deselect ''VTX'' when building your cloud BF firmware, then the BF Lua script will be stuck when trying to download the VTX tables. 

Maybe the Lua script could be updated to work with cloud builds without ''VTX''?

My workaround to use the BF Lua Script was to Backup my current BF Firmware, flash the same version just with ''VTX'' selected and load the backup.

Many Thanks

klutvott123, 2023-05-05:
try this https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases/tag/v20230313.7

closed 2023-06-12 by github-actions[bot]

mituritsyn, 2023-10-20:
> try this https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases/tag/v20230313.7

hi, this works for me, but I have this issue in latest release version
https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases/tag/v20231019.9

haslinghuis, 2023-10-20:
@atomgomba please have a look

@mituritsyn please provide support ID and an indication when the error occurs.

atomgomba, 2023-10-20:
I just tested [v20231019.9](https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases/tag/v20231019.9). I use TBS Sixty9 connected through CRSF  so "VTX" feature is not compiled in my firmware and hence no VTX table configured either. The radio used was a Tango 2. I deleted the dummy VTX table (lua file) from the SD card to force the script to try and download the VTX tables again. The result is that the Betaflight setup script works normally. We need more info to reproduce this.

mituritsyn, 2023-12-07:
> @atomgomba please have a look
> 
> @mituritsyn please provide support ID and an indication when the error occurs.

sorry, missed message
still reproduced. indication is "downloading vtx tables"
Id: "86080f48-f149-48ca-9c5e-b2f1a8ad25af

haslinghuis, 2024-01-01:
@mituritsyn please remove SCRIPTS/BF/VTX_TABLES/vtx_defaults.lua from SDC.