betaflight/betaflight-tx-lua-scripts #481: LiteRadio 3 Pro - Syntax Error

odixxx, 2023-06-27:
when running the Betaflight Setup script - Syntax Error
[image]

My setup:
Betaflight TX Lua Scripts 1.7.0
Betaflight 4.4.2
EdgeTX  v2.8.4

ps. on another radio X9D Plus everything works fine!

klutvott123, 2023-08-04:
it means it didn't find "scripts.lua". Did you download a zip file from here https://github.com/betaflight/betaflight-tx-lua-scripts/releases ?

Hofi, 2023-08-23:
I have the same error on a radiomaster zorro. I used the files from Release page.
[image]

klutvott123, 2023-08-24:
Can you check if /SCRIPTS/BF/COMPILE/scripts.lua actually exists on your SD card? Would be strange if it's not there.
this https://github.com/betaflight/betaflight-tx-lua-scripts/releases/download/1.7.0/betaflight-tx-lua-scripts_1.7.0.zip is the file with the compiled scripts

Hofi, 2023-08-24:
yes, it's there. i tried to copy the files multible times. I even tried reformatting the sd card.
[image]

Hofi, 2023-08-24:
I found the issue. I tried another sd card from sandisk and it works. I guess the sd card that sipped with the radio is garbage ...

closed 2023-10-01 by github-actions[bot]

raqso, 2023-10-02:
> when running the Betaflight Setup script - Syntax Error [image]
> 
> My setup: Betaflight TX Lua Scripts 1.7.0 Betaflight 4.4.2 EdgeTX v2.8.4
> 
> ps. on another radio X9D Plus everything works fine!

I wonder if you @odixxx  somehow managed to make this work on lr3 pro?

albka1986, 2023-11-05:
I also have the same issue, even though the files are present. 
Device: Readiomaster Boxer.

Hofi, 2023-11-05:
Have you tried another SD card? The SD card that came with the radio didn't work well for me. After changing SD card and copying files it should work.