betaflight/betaflight-tx-lua-scripts #486: Changing VTX settings from script causes issues with VTX Table

AndreaPro, 2023-09-01:
I am using Betaflight Lua Scripts 1.7.0 (I've even tried the nightly), Betaflight Configurator 10.9.0, Radiomaster TX16S with EdgeTX 2.8.5 and a BetaFPV Meteor 75 Pro updated to BF 4.4.2 and I am using ELRS 3.3.0 on BetaFPV Micro 1W TX and on the quad.
If I open the lua script and I adjust some settings in the VTX section and save them and I power off the quad and connect it to Betaflight configurator the VTX table is wiped completely and the settings are not stored. I've tried generating the lua script VTX table from the configurator and retrieving them from the quad but the result is the same. Pratically the VTX section of the script is unusable because after restart the VTX table is completely wiped.

haslinghuis, 2023-09-01:
Duplicate of https://github.com/betaflight/betaflight-tx-lua-scripts/issues/482

AndreaPro, 2023-09-01:
> Duplicate of #482

In my opinion is not a duplicate, it is a different issue. I have compiled the firmware with VTX and OSD already

klutvott123, 2023-09-01:
Does this happen when making the same changes in the betaflight configurator?

AndreaPro, 2023-09-01:
No, it happens only when making changes to VTX parameters using the lua
script from the radio. I've tried changing accelerometer trims and PID
profiles without issues

klutvott123, 2023-09-01:
can you try downloading the vtx tables again? https://github.com/betaflight/betaflight-tx-lua-scripts#setting-up-vtx-tables

AndreaPro, 2023-09-01:
Yes, I've tried multiple times both downloading them from the script on the
radio and even generating the lua script from the configurator and copied
it manually on the radio but the outcome is the same, as soon as I do one
change the VTX table is altered and becomes unusable

klutvott123, 2023-09-01:
That's strange. I'll have to see if I can reproduce

AndreaPro, 2023-09-01:
If there's something that I can do to troubleshoot the issue just let me
know

klutvott123, 2023-09-01:
Is it the setting of a specific parameter that causes the issue? Could you try 4.4.1 and maybe 4.4.0?

AndreaPro, 2023-09-01:
It happened when changing VTX power. I can try to go back to these versions
and let you know

AndreaPro, 2023-09-01:
I have downgraded to BF 4.4.0 that was the previous release available for
my board, and it happens the same. I've noticed one thing, that if I copy
the VTX table lua script generated by Betaflight inside the script it works
until I change a parameter, but if I download the VTX table from the quad
it downloads a different VTX table that is the wiped one that I found
inside the configurator. Attached there is the Lua script generated by
Betaflight configurator and the lua script that is created inside the radio
if I download the VTX table. And there is the VTX table json saved before
and after the issue.

klutvott123, 2023-09-02:
Seems like the lua scripts end up with the "empty" vtx table after downloading? We'll have to figure out why that happens then.
I just tested this with a betafpv meteor65 board with 4.4.2, and I'm not able to reproduce. Vtx table is correctly downloaded, and I can change all parameters.

klutvott123, 2023-09-02:
@AndreaPro Can you share a diff please?

AndreaPro, 2023-09-03:
> @AndreaPro Can you share a diff please?

For sure! I have tried both with the stock CLI provided by BetaFPV and with my custom settings but I have the same issue.

[CLI.zip](https://github.com/betaflight/betaflight-tx-lua-scripts/files/12504952/CLI.zip)

tomekste5, 2023-09-11:
Got the same issue if I download the vtx table or copy the lua script from betaflight to my controller, it overwrites my vtx table.

AndreaPro, 2023-09-11:
> Got the same issue if I download the vtx table or copy the lua script from betaflight to my controller, it overwrites my vtx table.

Could you please share more infos about radio firmware version, Betaflight version, script version, drone model, etc...?

tomekste5, 2023-09-11:
Oh ye sure sry,
Betaflight version: 4.3
Edge TX version: 2.8.0 (Radiomaster Boxer)
Betaflight lua script version: 1.7
Flight controller: Speedybee F405
VTX: TBS Unify Pro32 (https://www.team-blacksheep.com/vtx-table-for-betaflight.txt)

After I try to switch a setting, my VTX table on the flight controller gets overwritten to the same table u shared

klutvott123, 2023-09-12:
The only thing I can think of is that the payload somehow contains more bytes than it should, and this happens:
https://github.com/betaflight/betaflight/blob/99be07729963e4b5e5cb540555450bda28bd2781/src/main/msp/msp.c#L3380-L3407

It's strange because in the BF lua scripts we're only sending four bytes back to the FC

klutvott123, 2023-09-12:
@tomekste5 Which RX system are you using?

tomekste5, 2023-09-21:
I am using Express LRS version 3.3 2.4GHz and as a Receiver I use the Radiomaster RP1

AndreaPro, 2023-09-21:
> I am using Express LRS version 3.3 2.4GHz and as a Receiver I use the Radiomaster RP1

So we have in common ELRS 3.30 2.4GHz

klutvott123, 2023-09-22:
maybe try disabling vtx administrator in the elrs lua script.

AndreaPro, 2023-09-22:
> maybe try disabling vtx administrator in the elrs lua script.

I have tried VTX administrator in elrs lua script because the Betaflight script was destroying the VTX table, but I have never used it before discovering this issue

klutvott123, 2023-09-22:
try disabling it by setting "band" to "off". did this happen with earlier versions of elrs? the payload sent back from the betaflight lua scripts is only 4 bytes, but if more bytes are sent the vtx tables could be reset. not sure where those bytes are coming from. needs more investigating.

AndreaPro, 2023-09-23:
> try disabling it by setting "band" to "off". did this happen with earlier versions of elrs? the payload sent back from the betaflight lua scripts is only 4 bytes, but if more bytes are sent the vtx tables could be reset. not sure where those bytes are coming from. needs more investigating.

I have tried the Betaflight lua directly starting with ELRS 3.3.0. I will try disabling it but I am sure that the first time that I have noticed this issue I haven't enabled VTX admin yet

tomekste5, 2023-09-23:
> maybe try disabling vtx administrator in the elrs lua script.
I have never enabled VTX administrator

klutvott123, 2023-09-23:
If it's not too much hassle, could you guys please try ELRS 3.0? I'm not saying elrs is the issue, but we need to rule things out. As I said, the payload sent from the bf lua scripts is 4 bytes, but it could be that zero padding or something is added somewhere along the way to the FC.

tomekste5, 2023-09-28:
I couldn't find a target for the internal module of  Boxer radio for ExpressLRS v3.0

AndreaPro, 2023-09-30:
> If it's not too much hassle, could you guys please try ELRS 3.0? I'm not saying elrs is the issue, but we need to rule things out. As I said, the payload sent from the bf lua scripts is 4 bytes, but it could be that zero padding or something is added somewhere along the way to the FC.

I can try it and let you know

CBockelmann, 2023-10-12:
Have the same issue with
- Meteor65, Betaflight 4.42, ELRS 3.3 LBT
- SpeedyBee F405 Mini BLS 35A 20x20 Stack, Betaflight 4.42, ELRS 3.3 LBT (BetaFPV lite receiver) and TBS Pro 32 Unify nano.

Transmitter
- FrSky QX7 ACCST with BetaFPV ELRS 1W module at ELRS 3.3 (150Hz currently due to limits of the X7)

Configuration through betaflight works fine (OSD or Configurator), just activating the LUA script even with vtx tables from betflight configurator switches the vtx channel to another band. After that nothing works and a reboot is required to reset to the old settings.

zhornyak, 2023-10-18:
Hi 
Any update on this one?

Same issue here:
SpeedyBee F405 stack
BF 4.4.0
BF Lua 1.7.0
ELRS 2.4GHz v 3.3.0
Radiomaster TX12 Mk2 EdgeTX 2.9.1

Works on my Mobula8 without any  issues

AndreaPro, 2023-10-20:
> I can try it and let you know

Hello, finally I have been able to do some tests. I couldn't downgrade to ELRS 3.0 because there was no Target available for BETA FPV AIO RX. I've tried the following:

TX3.1.1 and RX3.1.1 = OK
TX3.2.1 and RX3.2.1 = OK
TX3.3.0 and RX 3.3.0 = BAD
TX 3.3.0 and RX 3.2.1 = OK

So the issue appears to happen with ELRS 3.3.0 on the RX side

klutvott123, 2023-10-20:
@AndreaPro Thanks for testing! Maybe open an issue on the ELRS github, or ask on their discord. I'm sure they have an explanation for it.

AndreaPro, 2023-10-21:
> @AndreaPro Thanks for testing! Maybe open an issue on the ELRS github, or ask on their discord. I'm sure they have an explanation for it.

[Done!](https://github.com/ExpressLRS/ExpressLRS/issues/2452#issue-1955368086)

0crap, 2023-10-22:
Same here. Latest stable v1.7.0 or nightly v20231020.10 wipes VTX tables on save.
Even reading the VTX tables goes wrong.
Reading gives:
Band 4
Channel 4
Power LV2
Pit Mode OFF
Protocol SA
Frequency 5800

While in fact my diff contains this:
# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWBAND  L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 3
vtxtable powervalues 14 20 26
vtxtable powerlabels 25 100 400

set vtx_band = 5
set vtx_channel = 3
set vtx_power = 3
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5732

0crap, 2023-10-22:
The <serialnumber>.lua on my SD card contains:

return {
    frequencyTable = {
        { 0, 0, 0, 0, 0, 0, 0, 0, },
        { 0, 0, 0, 0, 0, 0, 0, 0, },
        { 0, 0, 0, 0, 0, 0, 0, 0, },
        { 0, 0, 0, 0, 0, 0, 0, 0, },
        { 0, 0, 0, 0, 0, 0, 0, 0, },
        { 0, 0, 0, 0, 0, 0, 0, 0, },
    },
    frequenciesPerBand = 8,
    bandTable = { [0]="U", "1", "2", "3", "4", "5", "6", },
    powerTable = { "LV0", "LV1", "LV2", },
}

AndreaPro, 2023-10-22:
> Same here. Latest stable v1.7.0 or nightly v20231020.10 wipes VTX tables on save. Even reading the VTX tables goes wrong. Reading gives: Band 4 Channel 4 Power LV2 Pit Mode OFF Protocol SA Frequency [...]

Which drone and which ELRS version on TX and RX?

0crap, 2023-10-22:
@AndreaPro home made quad. ELRS 3.3.0 on both TX and RX.

AndreaPro, 2023-10-22:
> @AndreaPro home made quad. ELRS 3.3.0 on both TX and RX.

Which FC?

klutvott123, 2023-10-22:
it probably has something to do with the new MSP VTX code in ELRS 3.3.0

klutvott123, 2023-10-22:
updated to 3.3.0 now. All I have to do is start the bf lua and let it download vtx tables, and then they disappear from the FC

0crap, 2023-10-22:
> updated to 3.3.0 now. All I have to do is start the bf lua and let it download vtx tables, and then they disappear from the FC

That is not what happens here.
Only when I hit the "save page" button from the lua script my vtxtables is missing from the diff all.
Reading alone does not ruin my vtxtables.

klutvott123, 2023-10-22:
I just built elrs 3.3.0 without msp vtx support, and now it works fine, so there's definitely something going on in the RX code. downloading vtx tables works, and saving the vtx page works. vtx tables still there.

bloudman648, 2023-11-11:
this problem is still there on elrs 3.3.1...will there be a fix or should we build our own fw without msp support?
and does it only ereases the vtx table? i dont trust the lua script when changing on field pids or rates...

using edgetx 2.9.1 / tx16s mark ii
iflight blitz f7 stack & blitz 1.6w vtx (nazgul v3)

btw i am new to this fpv hobby :) 
started 2 weeks ago

klutvott123, 2023-11-12:
@bloudman648 It only affects vtx/vtx tables. Everything else should work.

There's not much we can do about this in the betaflight lua scripts. The script is working as it should and only sending the necessary bytes back when setting vtx parameters. The data is being tampered with somewhere else. 

You could report this issue here too https://github.com/ExpressLRS/ExpressLRS/issues/2452#issue-1955368086

jonathanoeijoeng, 2023-12-20:
I have exactly same issue with this.
I'm using Radiomaster boxer using elrs 3.3.0 dan rx radiomaster rp1 same elrs version.
Everytime I download vtx table, all reset just like this case.
I did edit vtx-table.lua and copy the correct table and power but problem still remain even I have restart both tx and rx

> https://github.com/betaflight/betaflight-tx-lua-scripts#setting-up-vtx-tables

jonathanoeijoeng, 2023-12-20:
Any luck with RX 3.3.1?

> Hello, finally I have been able to do some tests. I couldn't downgrade to ELRS 3.0 because there was no Target available for BETA FPV AIO RX. I've tried the following:
>
> TX3.1.1 and RX3.1.1 = OK [...]

AndreaPro, 2023-12-21:
> Any luck with RX 3.3.1?

I haven't tried it yet but if no one fixed it in the Betaflight or ELRS code repos I doubt that it will be fixed by itself

jonathanoeijoeng, 2023-12-21:
> ExpressLRS/ExpressLRS#2452 (comment)

you're right. I guess we stuck with 3.2.1 then

LoakAudio, 2024-01-06:
Perhaps a solution on the horizon ? https://github.com/ExpressLRS/ExpressLRS/pull/2525

haslinghuis, 2024-02-06:
Please confirm this is solved with https://github.com/ExpressLRS/ExpressLRS/pull/2525 included

AndreaPro, 2024-02-06:
> Please confirm this is solved with [ExpressLRS/ExpressLRS#2525](https://github.com/ExpressLRS/ExpressLRS/pull/2525) included

I haven't tested it yet because I was thinking that the merged commit will become available with the next ELRS update

haslinghuis, 2024-02-06:
I see we need to wait for 3.4 to come out.

haslinghuis, 2024-02-06:
Ah okay, just checked, we need version 3.4 to be released for testing.

LoakAudio, 2024-02-07:
> Ah okay, just checked, we need version 3.4 to be released for testing.

Tested today with `GIT Branch/Master` of ELRS and the penultimate `betaflight-tx-lua-scripts-nightlies`, VTX tables are loading correctly. No more issue for me.

haslinghuis, 2026-11-09:
Root cause found — this is an ExpressLRS receiver bug, not the Lua scripts.

In ELRS 3.3.x the receiver's MSPVTX module (meant for receivers with an integrated SPI VTX) is fed **every** MSP reply from the FC, and on receivers without an SPI VTX it is never disabled, so it stays in its initial `GET_VTX_TABLE_SIZE` state. When the Lua script reads `MSP_VTX_CONFIG` (MSPv1), ELRS parses the reply as its own MSPv2 reply with the payload offset by 3 bytes, decides the VTX table doesn't match, and sends `MSP_SET_VTX_CONFIG` with the "clear vtxtable" flag set and hard-coded 6 bands / 8 channels / 5 power levels, band 4 channel 4.

That matches the attachments exactly: `BAND1`..`BAND6` with letters `1`..`6`, all frequencies 0, power labels `LV0`..`LV4`. It also explains the channel jumping on just opening the script, and the extra bytes suspected earlier in this thread — they came from ELRS, not the script (which sends 4 bytes).

Fixed upstream in **ELRS 3.4.0** (ExpressLRS/ExpressLRS#2525, "Disable processing MSPVTX commands if there is not SPIVTX"). 3.3.1 and 3.3.2 are still affected.

To recover:
1. Update the receiver to ELRS 3.4.0 or newer.
2. Load your correct VTX table in the Configurator and save.
3. Re-download the VTX tables from the Lua script.

Closing as an upstream issue.

closed 2026-11-09 by haslinghuis