betaflight/betaflight-tx-lua-scripts #487: Taranis QX7 stuck on Initialising

ExoDuCe, 2023-09-05:
Hey everybody,

my Radio is stuck on initialising with the following hardware and software:
Taranis QX7 ACCST (OpenTX 2.3.14)
R-XSR Receiver (ACCST D16 v2.1.1(F.Port Version)
--> therefore I use BF Lua Script 1.5.0 (F Port 2.1.1 is not supports with 1.6/1.7)
Betaflight is on 4.3.2 BTFL version

I can discover and receive sensor data in the taranis Telemetry tab, but the BF Setup script always gets stuck on initializing.

Would love some help :)

klutvott123, 2023-09-07:
2.1.1 is buggy. try 2.1.0. The latest bf lua release will also support older betaflight versions

ExoDuCe, 2023-09-08:
I‘ve also tried that right now also not working
Current setup:
2.1.0 r-xsr with s.port (not f.port)
BF 4.4

klutvott123, 2023-09-08:
@ExoDuCe So were you using f.port or sbus/s.port with 2.1.1?

ExoDuCe, 2023-09-08:
Originally I was using F.Port 2.1.1 but with BF Lua Script 1.5
Bc. further versions weren't able to use the F.Port 2.1.1

klutvott123, 2023-09-09:
1.6 and 1.7 supports 2.1.1, but that doesn't help when 2.1.1 has bugs, and that's why we recommend not using it. Please try 2.1.0 Fport firmware. If you want to use 2.1.1 you have to contact Frsky and get them to fix it.
Are you seeing ALL telemetry sensors on your radio, or just the ones coming directly from the rx?

closed 2023-10-18 by github-actions[bot]