betaflight/betaflight-tx-lua-scripts #489: Radiomaster Boxer EdgeTX 2.9.1 - Error on script compilation

marcinczeczko, 2023-10-02:
1. Problem description
On a brand new Radiomaster Boxer with internal ELRS with Edge TX 2.9.1 i installed BF Lua by copying to SDcard. Once I start script, I'm getting compilation error: "Syntax error, scripts.lua, "bt") error a" File not found

2. Ways to reproduce
- Radiomaster Boxer with ELRS
- EdgeTX 2.9.1.
- Install BF Lua
- Try to run BF Lua

Attached is recording of what's actually happening

https://github.com/betaflight/betaflight-tx-lua-scripts/assets/4218395/ad97499a-78a3-41bb-9270-1aef61ea6254

klutvott123, 2023-10-02:
Please try another sd card

marcinczeczko, 2023-10-03:
@klutvott123 thanks ! I replaced sd card and now it compiles just fine. Looks like the sd card that came together with radio is not a good quality one :(

closed 2023-10-03 by marcinczeczko