betaflight/betaflight-tx-lua-scripts #491: Convert betaflight-tx-lua to Inav

druckgott, 2023-11-03:
## If you believe there is an issue with the firmware itself please follow these steps:
I use this script very often at betaflight and it´s realy nice for PID Tuning an do so on.
For my funjet I need now inav for flying fixed wing and I convert the script that it alrady start and have no error with Inav (at first I only want to use it for pid tuning)

I renamed all things and so on. But there is one topic which is different in Inav the MSP values are extending the value 255.

I can recive values from here:
https://github.com/iNavFlight/inav/blob/master/src/main/msp/msp_protocol.h

But Inav also use this area starting by: 0x2000
https://github.com/iNavFlight/inav/blob/master/src/main/msp/msp_protocol_v2_inav.h

I now add a page wher the value: 0x2000 (or in decimal I also tested it) have to load there comes nothing.

The Problem is I do not have this huge experince to change the msp code to also recive values and send values to this adresses.

Maybe somebody from the betaflight team can help me to get this cool script working also on Inav as a first version.
My current Version is here:
https://github.com/druckgott/inav-tx-lua-scripts

Hopefully here is one person who can help me also if it´s inav ;-)

klutvott123, 2023-11-12:
Currently the betaflight lua scripts only support MSPV1, but MSPV2 support should be pretty straight forward to implement. The problem is that it won't work with frsky or ghost as the packets are too small, but for CRSF it's doable.

Have a look at this https://github.com/betaflight/betaflight-tx-lua-scripts/blob/master/src/SCRIPTS/BF/MSP/common.lua 

I'll have a look at it too when I have the time.

druckgott, 2023-11-26:
Hi thx. for your link. I have no Idea how to implement this. I compared it to here:
https://github.com/fpv-wtf/msp-osd/pull/140/files
But it´s to complex for me ;-)

klutvott123, 2023-11-28:
@druckgott I just had a look at the Inav code, and it doesn't have support for mspv2 over telemetry like betaflight does so you would have to update the firmware for it to work too. I have mspv2 working on the bf lua scripts right now, but the problem is still that it will only work with the CRSF protocol. 

anyway here's something that works https://github.com/betaflight/betaflight-tx-lua-scripts/compare/master...klutvott123:betaflight-tx-lua-scripts:msp_v2

klutvott123, 2023-11-28:
also I would recommend basing your INAV version on an actual fork of the BF scripts to preserve commit history 🙂

druckgott, 2023-11-28:
Ah im yes i can do this, makes sence. So i will have a Look forward thx for your help very nice.

druckgott, 2023-12-01:
Ok I did a fork again and put in your things thx for your help. Now there needs to be inav updated

klutvott123, 2023-12-01:
look at these
https://github.com/betaflight/betaflight/pull/11112
https://github.com/betaflight/betaflight/pull/11131

druckgott, 2023-12-01:
Thx i will have a look

druckgott, 2023-12-02:
Its done, but I do not know how to debug it:
https://github.com/iNavFlight/inav/pull/9528

klutvott123, 2023-12-02:
There may be differences between inav and bf to begin with. not sure, but you should double check. also have a closer look at the second PR I linked to. The first PR introduced a shared msp buffer, but that didn't work too well so it was removed in the second PR. You have two mspSerialOutBuf in your PR

druckgott, 2023-12-02:
I have expresslrs protocoll so crfs

Ok i will Check this with the double Definition later

closed 2024-01-10 by github-actions[bot]