betaflight/betaflight-tx-lua-scripts #497: VTx Band - Now numbers instead of Letters

SpeedyPizzaDrones, 2024-02-05:
- Use LUA 1.7.0
- Go to VTx settings
- BAND: Scroll all bands and you'll find U;1;2;3;4;5 instead of A;B;E;F;R.

Having numbers instead of letters is way less intuitive

haslinghuis, 2024-03-07:
@klutvott123 @atomgomba did you check or can verify this?

klutvott123, 2024-03-07:
please post the screenshots here too

atomgomba, 2024-03-08:
I can't find the original message on discord (on mobile currently), but I have advised the OP to create an issue. @SpeedyPizzaDrones could you please post the screenshots here as well?

atomgomba, 2024-03-08:
> @atomgomba If you know what causes this, please tell :)

No, but the protocol was Trump IIRC

haslinghuis, 2026-11-09:
Duplicate of #486. Bands showing as `U;1;2;3;4;5` means the VTX table on the flight controller itself has been wiped (cleared bands get the letters `1`..`6`); the script just shows what the FC reports.

That wipe is caused by ELRS 3.3.x receivers, fixed in ELRS 3.4.0 — see #486 for details. To recover: update the receiver to ELRS 3.4.0 or newer, load your correct VTX table in the Configurator and save, then re-download the VTX tables from the Lua script.

closed 2026-11-09 by haslinghuis