betaflight/betaflight-tx-lua-scripts #503: Cannot change VTX settings.  Page does not seem to respond.

stephenwilley, 2024-03-19:
I'll start this by saying I'm running Betaflight 4.5RC2.  If the answer here is 'we don't support it 'till it's released', that's fine with me.

The VTX tab is simply not editable.  It opens and stays with band at U, channel 0 etc. etc.
I *can* change channels, bands, etc. using the OSD in goggles with stick controls so SmartAudio control is working.  This change is not reflected in the tab on the radio.
The option to redownload the VTX table does work (I have checked the .lua on the SD card and it has correctly pulled the values)
I have tried setting a specific frequency on the Betaflight configurator as I saw that suggested somewhere.  No difference.

Other tabs are editable and can be saved (rates for example).

What hardware / software you are using:

Taranis X9D+ running 2.3.12
Betaflight 4.5RC2 on a CLRacingF4S board connected to an X4R-SB running SBUS and a TBS Unify Race Pro HV with SmartAudio

haslinghuis, 2024-03-19:
No mention about lua script version used. Maybe try https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases

stephenwilley, 2024-03-19:
Ah sorry - missed the most obvious version to mention there.  I'm running 1.7.0
I can definitely try the nightlies

stephenwilley, 2024-03-20:
No change with nightlies I'm afraid.  Did a full delete of all the compiled file and redid the Setup script from the start.

VTX tables did download again.  I've taken the lua from the SD Card and just changed the extension to TXT so I could attach it here in case that's useful.
[001f00263436471935303038.txt](https://github.com/betaflight/betaflight-tx-lua-scripts/files/14658132/001f00263436471935303038.txt)

The tab shows:
BAND U
Channel 0
Power 200
Pit Mode: Off
Protocol: SA
Frequency: 26

I can navigate up and down them but the ENTER key does not start the cursor flashing.  A long press of ENTER and choosing save page returns "Script syntax error ?:0 attempt to compare number with nil" which makes sense but figured I'd add it for completeness.

I'll take a peek at the code and see if I can spot any conditions that would make the page non-editable.

klutvott123, 2024-03-20:
did it work before you flashed 4.5 RC2?

stephenwilley, 2024-03-20:
4.5RC2 was the first time I'd flashed anything since 4.0 (I had my quads under the stairs for a while) so I bumped everything to latest versions and had never used these lua scripts before.

If there's a 'most reliable' BF version you'd like me to try, I can do a diff backup of my setup, save-to-file the VTX setup and then downgrade BF and do a minimal setup to test it.

klutvott123, 2024-03-20:
4.4 is fine. also try setting band to something other than 0 in the configurator and save. see if it works then

stephenwilley, 2024-03-20:
Ah, there we go.

set vtx_band = 5
set vtx_channel = 1
save

did the trick.  Thank you.  I had previously tried using the 'Selected Mode' panel of the configurator to enter a frequency directly, but, it didn't have the options to choose Band and Channel until I'd manually set those in the CLI.

Not sure if this is something that needs to be 'fixed' or just maybe add a recommendation to the README?

Appreciate your help getting this working.  Really nice menu system :-)

klutvott123, 2024-03-21:
@stephenwilley Nice! I think it's always been like this, but we could probably improve this.

haslinghuis, 2026-11-09:
Fix proposed in #553: the VTX page discarded the flight controller's reply whenever channel was 0, which is always the case in user-frequency mode (`vtx_band = 0`), leaving every field read-only and causing the save error. Testing on a radio with `vtx_band = 0` would be appreciated.