betaflight/betaflight-tx-lua-scripts #510: Very slow update w/ TBS Crossfire

invicta-nz, 2024-09-07:
Issue: Updates to values are very, very slow (like 3-5 seconds) when trying to update BF settings using EdgeTX and TBS receiver 

Platform: Running most recent EdgeTX and TBS crossfire firmwares as of 8/11/2024. Running most recent lua script as well. Radiomaster Boxter radio. 

Basically the interface works, but functionally not really as every change in the menu requires 3-5 seconds to update. The commands are obviously sent and received, but with such a delay as to be unusable.

haslinghuis, 2024-09-07:
Did you try with https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies ?

invicta-nz, 2024-09-08:
No change with most recent dev build, still unusably laggy and slow to update. 

Is updating and reverting correctly accomplished by copying the two directories and overwritting files? No other steps required?

pfeerick, 2024-09-09:
Yes, that is all that should be required. Note that Crossfire has a very slow telemetry rate... 150Hz at best... so it is possible this is simply the link speed. Tracer has a higher rate at 250hz.

invicta-nz, 2024-09-09:
But this is more like 0.2hz update. Have to wonder if there is something else going on with my setup as I can't see this being usable for anyone. Do you know if anyone is testing with crossfire?

invicta-nz, 2024-09-12:
BF 4.5.

Can try flashing to 4.4 this weekend and see if it changes anything

closed 2024-10-21 by github-actions[bot]