betaflight/betaflight-tx-lua-scripts #512: Ethos

robthomson, 2024-10-18:
Just putting this out there in the hope of a contact from relevant devs.

I am the guy who has built the lua for ethos/rotorflight.

https://github.com/robthomson/RF2ETHOS

Do you guys want me to do a verson that supports betaflight?

Its pretty simple for me to do - just a question of if you want it?

haslinghuis, 2024-10-18:
@klutvott123

robthomson, 2024-10-21:
Well.. Remarkable amount of enthusiasm :)

No issue.. Will just do it anyway.

haslinghuis, 2024-10-21:
Will try on discord.

robthomson, 2024-11-01:
So this has now started.
[image]

robthomson, 2024-11-01:
Worth noting.

I am doing this port for the community.   I have not flown quads in years now - so no real gain other than just getting it done because nobody else is bothering.

I would ideally like to get the guts of it done - then hand off the source to another BF orientated developer.    

All up...  I am a little dissapointed in the complete lack of engagement from the BF team in even so much as contacting back when a developer offers to do something.    Kind of sad really.

pfeerick, 2024-11-06:
In addition to this being a community project itself, I think that is probably a rather telling indication of the percentage of the user base of Ethos running BF 🤭 Plus, since I would be willing to be bet none of the BF developers have ETHOS radios, and they would be reliant on someone else to do any testing, that would reduce any enthusiasm (aka time/resources to devote to this) somewhat also. Regardless of the project and handset OS, Lua developers seem to be few and far between ... and it can be quite some time before anything happens. 😮😢

robthomson, 2024-11-10:
I have never needed them to do anything.

Round 1 is already done and working.

https://github.com/robthomson/betaflight-ethos-scripts-simple

klutvott123, 2024-12-08:
@robthomson Thank you for the effort :)
Unfortunately I don't think anyone on the BF team has the equipment to test or maintain this.

robthomson, 2024-12-08:
Such is life.

More than happy to put you in contact with somebody who can help you in that regard - but it does probably mean that somebody using bf needs to be interested in doing it.

I have not flown quads for years now so not likely to be myself.   Maybe somebody else with interest will pick it up.

In the mean time..   Focusing my attention on the rotorflight project.

closed 2025-01-16 by github-actions[bot]