betaflight/betaflight-tx-lua-scripts #513: No option or details on how to save changes

bigjoncoop, 2024-11-15:
**Betaflight TX Lua Scripts**: 1.7.0
**Firmware**: OpenTX - Latest Version 
**Hardware**: Seems to be "at least" radios that have B&W non-touch displays. (e.g. Taranis QX7, X9D, X9) 

**Main Issue**:
While in the Lua Script making changes, there is no Button/Selection or any details on how to save said changes... This makes the user automatically assume that the changes are auto saved, which of course there not. After trying multiple times and getting frustrated, users _rightfully_ assume that the Lua script is broken. 

This originally happened to me a while back and after trying multiple times and changes not being saved, I assumed that there was an issue with the Lua Script... Then months later after I stopped using it, I decided to factory reset my radio and start from scratch. To my surprise after setting everything back up and reinstalling the Lua script, the problem still persisted. Then after doing a Google search I learned that in order for changes to be saved you have to long press the "Enter button" and select "Save Page".

### Over the past few years at meetups, I've had multiple people mention they were having the same exact issue. After showing them what the problem was they all basically said the same thing, why doesn't it have a save button in the Lua script or mention it in the Lua script menu, etc...

klutvott123, 2024-12-08:
There's a user manual :) https://github.com/betaflight/betaflight-tx-lua-scripts#how-to-use

pfeerick, 2024-12-13:
I don't mean to add unhelpful "noise", but while that is true... it would be nice if the existence of the "long press" menu was somehow brought to a new users attention and/or the "unsaved" state of settings changes... The later is probably easier to indicate, but not sure how to make the former visible outside of a brief oneshot "first run" experience in the lua script that hints at the UI conventions and any "hidden" shortcuts.

closed 2025-01-21 by github-actions[bot]